trezor/trezor-firmware #1543: Metamask+Trezor fail to sign EIP-712 message

drortirosh, 2021-03-23:
**Describe the bug**

Trying to use a dapp that requires EIP-712 (ctf-react.opengsn.org)
Works with Metamask account, but when using metamask with Trezor wallet we get an error message:
`MetaMask Message Signature: Error: Not supported on this device`

**Firmware version and revision**
latest firmware: 1.9.4
Describe firmware version and revision you are using when encountering the bug.

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Metamask
 - OS: macOS
 - Browser chrome

**To Reproduce**
Steps to reproduce the behavior:
1. Go to ctf-react.opengsn.org
2. Connect your metamask to the dapp.
3. Click on "Capture the Flag"
4. see the "sign" message. click "Sign"
5. see the above error message.
6. (note that if you use with metamask "native" account, not connected to trezor, it does work as expected.)

**Expected behavior**
To sign properly

prusnak, 2021-03-23:
No-one implemented EIP-712 signing on Trezor yet.

Duplicate of https://github.com/trezor/trezor-firmware/issues/1272

closed 2021-03-23 by prusnak

prusnak, 2021-12-06:
EIP-712 for Trezor One is being developed here and will be hopefully released in January: https://github.com/trezor/trezor-firmware/pull/1970

alisinabh, 2021-12-06:
@prusnak @matejcik Due to the lock in #1970 I cannot reply. Can you unblock me somehow? Many thanks.
(Sorry I wrote here)