trezor/trezor-firmware #1553: Legacy self complied firmware doesn't not get jump from bootloader.

vladT0, 2021-03-29:
Self complied firmware give me the error of unofficial firmware detected. The "Unofficial firmware" warning and the hash confirmation is expected. I always compile with MEMORY_PROTECT=0 and my bootloader is not locked.
I pressed button and I have message about "compare fingerprint"). Pressed button again and nothing happened.
MPU did not start to execute trezor.c "main" function. I investigated situation and found, the reason is
in function from util.h:
`jump_to_firmware(const vector_table_t *ivt, int trust)`
I changed flag "trust as true" and firmware successfully started. But it is not correct way, because soon later I will get a "hard fault detected" instead. Did I do something wrong?

Any help will be appreciated.

Bootloader 1.10

tsusanka, 2021-03-29:
@vladT0 could you please try to set `MEMORY_PROTECT=1` when building bootloader and `MEMORY_PROTECT=0` when building firmware as documented [here](
https://github.com/trezor/trezor-firmware/blob/master/docs/legacy/index.md#combining-bootloader-and-firmware-with-various-memory_protect-settings-signedunsigned)?

If that does not help, could you please check out tag `legacy/v1.9.4` and try there? We are about to release a new bootloader (1.10 that's what is in master) which is not fully tested yet so there can also be something unexpected.

vladT0, 2021-03-29:
I have developing hardware based on STM32F4DEV boards and would not like to lock it and block :) Is it save enough ?

tsusanka, 2021-03-29:
cc @prusnak @hiviah

prusnak, 2021-03-29:
> I have developing hardware based on STM32F4DEV boards and would not like to lock it and block :) Is it save enough ?

Other option is to use this row from the table:

| Bootloader MEMORY_PROTECT | Firmware MEMORY_PROTECT | Is firmware officially signed? | Result                                                                                     |
| ------------------------- | ----------------------- | ------------------------------ | ------------------------------------------------------------------------------------------ |
|  0                        |  1                      | no                             | works, but don't forget to comment out `check_bootloader`, otherwise it'll get overwritten |

hiviah, 2021-03-29:
I think the above (bootloader `MEMORY_PROTECT=0` and firmware `MEMORY_PROTECT=1`)  is the best combination which I used for development. Just be careful you need to comment out the `check_bootloader` code in firmware, otherwise it will overwrite your unlocked bootloader and then the official bootloader that replaced it will lock out JTAG/SWD.

vladT0, 2021-03-29:
I found some posts from year 2018 where @jhoenicke  also recommended to comment `timer_init`
https://github.com/trezor/trezor-mcu/issues/346#issuecomment-387036589
is it relevant to the actual version of bootloader too ?

prusnak, 2021-03-29:
> is it relevant to the actual version of bootloader too ?

No, the new firmware runs `timer_init` only if running in priviliged mode (signed firmware)

e239, 2021-04-05:
###

e239, 2021-04-05:
hello world

prusnak, 2021-04-06:
So I guess the issue is resolved. If there are any further questions, feel free to ask Vlad

closed 2021-04-06 by prusnak

vladT0, 2021-04-06:
Thank you very much for your help, configuration  with MEMORY_PROTECT = 0 for Bootloader and MEMORY_PROTECT = 1 for FIRMWARE is working.