trezor/trezor-firmware #1556: Does Trezor Plan To Implement EIP2333 With A Firmware Update?

AdamSHurwitz, 2021-03-31:
**Question — Are there plans from the Trezor team to support _[EIP2333][1]_ through a firmware update?**

From the comments in _[Trezor and Ethereum 2.0][2]_ in _[r/TREZOR][3]_ (3/29/21) a community member stated Trezor will not implement EIP2333, at least in the short-term.

> Trezor needs to implement EIP2333 which it hasn't nor has any plans to…

**Additional context**
The utility of EIP2333 is for hardware wallets to use their existing keys to interact with third-party Eth2 validator node services.

- Third-party Eth2 validator node services create the validator key.
- Hardware wallets create the withdrawal key from the third-party based on the wallet's existing seed.

In contrast, hardware wallets without EIP2333 would make the initial deposit to a third-party Eth2 validator service, and a brand new withdrawal seed will be created that the user will securely store and manage in addition to their original wallet's seed.

[1]: https://eips.ethereum.org/EIPS/eip-2333
[2]: https://www.reddit.com/r/TREZOR/comments/k1j5vm/trezor_and_ethereum_20/gsqw6pt?utm_source=share&utm_medium=web2x&context=1
[3]: https://www.reddit.com/r/TREZOR/

closed 2021-03-31 by prusnak

AdamSHurwitz, 2021-03-31:
Thank you for the feedback @prusnak! "Someone" I estimate meaning an Eth2 third-party service provider implements EIP2333 appropriately?

prusnak, 2021-03-31:
> "Someone" I estimate meaning an Eth2 third-party service provider implements EIP2333 appropriately?

Yes