trezor/trezor-firmware #1558: Pre-staking INJ, signing with error: Forbidden Key Path

kccha75, 2021-04-02:
**Describe the bug**
When attempting to pre-stake INJ, cannot Approve transfer of ERC20 tokens into Injective in the Kovan Test Network. See walkthrough here: https://blog.injectiveprotocol.com/equinox-staking-walkthrough/ at Step 4.

Error encountered: Error: Forbidden Key Path, during transaction and data signing.

**Firmware version and revision**
2.3.6

**Desktop/smartphone setup (please complete the following information):**
 - Metamask
 - Windows
 - Chrome

prusnak, 2021-04-02:
You are using testnet and testnets need to use `m/44'/1'` paths instead of `m/44'/60'`. This is a bug in the wallet you are using that does not respect BIP-44.

closed 2021-04-02 by prusnak

kccha75, 2021-04-02:
Sorry I'm quite new to this, but is there anything I can do about it?