trezor/trezor-firmware #1563: Encountered Invalid prevhash

liyanhrxy, 2021-04-07:
**Describe the bug**
when I use trezor one to signing my BTC transaction, I got  the error message `Encountered Invalid prevhash` ? what's it means ?? 
**Firmware version and revision**
1.9.8
**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: [Trezor Suite]
 - OS: [macOS]
 
**To Reproduce**
hard to reproduce

prusnak, 2021-04-07:
> 1.9.5

There is no such firmware relase 1.9.5 - where did you install this firmware from?

liyanhrxy, 2021-04-07:
ah, I'm sorry, it is 1.9.4

prusnak, 2021-04-07:
Which version of Suite are you running?

If it's not the latest (21.3.2) please install latest Trezor Suite from https://suite.trezor.io

liyanhrxy, 2021-04-07:
> If it's not the latest (21.3.2) please install latest Trezor Suite from https://suite.trezor.io

it's newest

liyanhrxy, 2021-04-09:
when this error(`Encountered Invalid prevhash`) occurred, what's mean? I have seen this error in many place, it isn't new!!! what reason cause this?