trezor/trezor-firmware #1566: core for T1 not building due to USB config

mmilata, 2021-04-12:
#1539 broke core build for Trezor 1: https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/jobs/1170543371

Tried switching to high speed like TT, build passes but after booting the device is not visible on USB bus:
```diff
--- a/core/embed/trezorhal/stm32f4xx_hal_conf.h
+++ b/core/embed/trezorhal/stm32f4xx_hal_conf.h
@@ -52,7 +52,8 @@
 #define USE_USB_HS
 #define USE_USB_HS_IN_FS
 #elif STM32F405xx
-#define USE_USB_FS
+#define USE_USB_HS
+#define USE_USB_HS_IN_FS
 #else
 #error Unsupported MCU
 #endif
```

Note that currently the T1 core UI is in such a disrepair that the following patch is needed to get it to the point where usb is initialized: c1fb285d8f22412f35433f1d502bd9ff78842d35. Also device with unlocked bootloader is needed to test it.

cc @mcudev - any ideas what to look for?

mcudev, 2021-04-12:
interesting. i wasn't aware of the T1 dependency on this and so i only tested TT.

the first thing that i did just now was to look up the usb hs peripheral specs for the STM32F205RET6 in RM0033 section 30.2.3 and the STM32F427VIT6 in RM0090 section 35.2.3 to compare them. they look compatible. both docs also list a 4-Kbyte data RAM and i believe support for hs in fs mode. so that is all promising as far as being able to get this working the same on both devices. honestly, i'll need to get current on the T1 code. just so i know what to work on, how high of a priority (how severe) is this? do we need a fix now, a day, a week, etc...?

mmilata, 2021-04-13:
This is part of the effort to port the micropython-based FW to T1, there's no dependency with the current production firmware. There are other major things we need to do before the port is complete so IMO the priority is "nice-to-have" since we can probably bring back the `#if defined(USE_USB_FS)` block for now and live with less endpoints.

mcudev, 2021-04-13:
Great, thanks for the info and the report!
I can reproduce this with `make build_firmware BITCOIN_ONLY=1 TREZOR_MODEL=1` and will work on it relatively soon. I have unlocked devices to test it.

mcudev, 2021-04-13:
I think that we might be stuck using the FS with the T1.
The T1 is wired into the USB_OTG_FS peripheral and not the HS like the TT is.
I'm going by the schematics at
https://raw.githubusercontent.com/trezor/trezor-hardware/master/electronics/trezor_one/trezor_v1.1.sch.png
and
https://raw.githubusercontent.com/trezor/trezor-core/master/docs/hardware/schematic.png

My T1 dev kit boards are wired the same as the T1 devices to PA11 and PA12
https://www.waveshare.com/w/upload/0/05/CorexxxR-Schematic.pdf

PA11 and PA12 map to OTG_FS_DM and OTG_FS_DP
while
PB14 and PB15 are OTG_HS_DM and OTG_HS_DP

So, to support T1 I think I have to add the FS code back to core/embed/trezorhal/usbd_conf.c
No big deal. Just a limitation that I don't think we can work around in software unless I'm missing something.

matejcik, 2021-04-13:
> So, to support T1 I think I have to add the FS code back to core/embed/trezorhal/usbd_conf.c

does that mean that T1 won't be able to have 4 endpoints, or is that a separate issue?

mmilata, 2021-04-13:
> does that mean that T1 won't be able to have 4 endpoints, or is that a separate issue?

Yes, at least if I understand [AN4879](https://www.st.com/content/ccc/resource/technical/document/application_note/group0/0b/10/63/76/87/7a/47/4b/DM00296349/files/DM00296349.pdf/jcr:content/translations/en.DM00296349.pdf), table 5 correctly.

mcudev, 2021-04-13:
Tentatively, if my understanding is correct, then I don't think that the T1 will be able to support all of the USB interfaces (wire, debug, webauthn, and vcp) at the same time. The OTG_FS peripheral only has a 1.25 Kbyte RAM and 3 instead of 5 IN/OUT endpoints (from section 29.2.3 in RM0033 for T1 and section 35.2.3 in RM0090 for TT).
The thing that makes me wonder if I can still somehow configure the T1 to use those pins with the OTG_HS in FS mode is that the datasheet has "OTG_FS/ OTG_HS" in the column heading of the alternat function mapping table. That's why I was wondering if there is something that I'm missing. It's worth a little time for me to tinker.

mcudev, 2021-04-13:
I think the T1 will be limited as I originally thought. I put up some code to try that worked for me at https://github.com/trezor/trezor-firmware/compare/master...mcudev:t1_usb. I used @mmilata's UI patch, c1fb285d8f22412f35433f1d502bd9ff78842d35, with this too.

I thought about updating `core/src/usb.py` also (might still have to), but when I build with `make clean vendor build_firmware BITCOIN_ONLY=0 TREZOR_MODEL=1` there is too much code to fit into flash.
A debug build seems to work too: `make clean vendor build_firmware BITCOIN_ONLY=1 TREZOR_MODEL=1 PYOPT=0`
There's probably something with the VCP interface that doesn't work that I didn't test that would make us make the additional changes to `core/src/usb.py`. But, I didn't find it yet.

mmilata, 2021-04-14:
Thanks for resolving this quickly @mcudev! I've opened #1569 with your changes. The size of `BITCOIN_ONLY=0` is a known issue.

PR #1569 (2021-04-14): core/embed/usb: add back usb fs code for T1 and redo fifo sizing
Fixes #1566 - core T1 build. See also #1539.
closed 2021-04-14 by prusnak