trezor/trezor-firmware #1572: Uniswap says "soon" at https://trezor.io/coins/ while it!s ERC20 token

yohaaan, 2021-04-15:
**Issue:** Uniswap is ERC20 token, hence supported by default, it's displayed with status "soon"
It's already recognized in Suite and can be added:
[image]
URL: https://trezor.io/coins/#UNI

**Suggested solution:** change status to "Supported" and add available ETH wallet options.

yohaaan, 2021-04-21:
Same goes for GRT https://trezor.io/coins/#GRT

yohaaan, 2021-04-22:
And for AMP https://trezor.io/coins/#AMP

matejcik, 2021-04-27:
tio/coins only show tokens as "supported" if Trezor will show the correct symbol/decimals.

support for UNI, GRT, AMP is tracked from v1.9.5/2.3.7, which is not out yet. So Trezor will show "UNKN" as the symbol. Hence, not technically supported.

closed 2021-04-27 by matejcik