trezor/trezor-firmware #1574: Add support for Polkadot (DOT)

jziggas, 2021-04-19:
**Is your feature request related to a problem? Please describe.**
As a [Polkadot](https://polkadot.network/) (DOT) user I would like to store DOT coins on my Trezor Model T.

**Describe the solution you'd like**
The Trezor firmware supports 3rd party wallets for Polkadot (#&#8203;8  by market cap).

**Describe alternatives you've considered**
[Ledger Wallet](https://support.ledger.com/hc/en-us/articles/360016289919-Polkadot-DOT-)

**Additional context**
https://polkadot.network/

prusnak, 2021-04-19:
Our current product goal is to unite what we support in firmware and in Trezor Suite and since firmware is way ahead of Suite we want to pause for a bit, implement the remaining coins into Suite and then consider adding new coins. Thank you for your understanding.

closed 2021-04-19 by prusnak

jziggas, 2021-04-19:
I don't see a milestone for that in the trezor-suite repository. Is there a link or ticket that tracks feature parity between trezor-suite and trezor-firmware?

prusnak, 2021-04-19:
Here is the public roadmap for the Suite - https://github.com/orgs/trezor/projects/28

Once everything from the Upcoming is cleared we might consider adding new coins.

jziggas, 2021-04-19:
Ok thanks. So for posterity, only after these 15 issues are resolved would any new coins be considered:

https://github.com/trezor/trezor-suite/issues/3550

https://github.com/trezor/trezor-suite/issues/2770

https://github.com/trezor/trezor-suite/issues/3344

https://github.com/trezor/trezor-suite/issues/2736

https://github.com/trezor/trezor-suite/issues/2838

https://github.com/trezor/trezor-suite/issues/2740

https://github.com/trezor/trezor-suite/issues/2745

https://github.com/trezor/trezor-suite/issues/2918

https://github.com/trezor/trezor-suite/issues/2959

https://github.com/trezor/trezor-suite/issues/3002

https://github.com/trezor/trezor-suite/issues/3004

https://github.com/trezor/trezor-firmware/issues/1404

https://github.com/trezor/trezor-suite/issues/3016

https://github.com/trezor/trezor-suite/issues/3166

shessam, 2021-08-04:
Pleaaase add Polkadot (DOT) on Trezor Model T.  
When :'(

NoodleSploder, 2021-08-12:
Adding Cardano but not PolkaDot?  this is insanity.

h4x3rotab, 2021-09-05:
@prusnak Now the dashboard doesn't have an "upcoming" column. What's the backlog now?

Btw, if someone contributes a PR, will Trezor team be willing to review?

JiechengZhao, 2022-01-18:
+1 dot support needed.

VadimChiritsa, 2022-02-27:
Could you please provide ETA for this?

hynek-jina, 2022-02-28:
@VadimChiritsa 
There is no ETA for Polkadot support at this moment.

timemaster5, 2022-03-11:
+1 I also vote for DOT

Sm1le291, 2022-03-15:
Add support for dot please

meehow, 2022-03-17:
@prusnak we (@weareseba) are willing to offer you some reasonable financial incentive or sponsor development of DOT and SOL in Trezor T wallet. I filled out a [from](https://trezor.io/enterprise/) on your website.

KarpiuOPP, 2022-04-03:
Also waiting for DOT support :)

anttivermas, 2022-04-14:
I am waiting for it also.

JiechengZhao, 2022-04-15:
@prusnak Could you reopen the issue?

sefechit, 2022-04-15:
+1 on DOT :(

faisalthaheem, 2022-04-18:
+1 for dot

Valcyclovir, 2022-04-20:
+1 for dot, ledger does it