trezor/trezor-firmware #1576: Firmware 1.9.4 broke support for Firo /XZC/ and ZnodeMasternodetool - Firmware 1.9.3 work without issues

jTd7bPLFb, 2021-04-20:
Not possible to manage FIRO masternodes with firmware 1.9.4 and ZnodeMasternodeTool.

Not possible to sign transactions or send funds.
Problem is that transaction confirmation just flashes briefly in Trezor display and can not be confirmed with Trezor button.

Version 1.9.3 of Trezor Firmware works without issues though.

Testen on MacOS and Windows 10. Trezor fw 1.9.4 and 1.9.3.

prusnak, 2021-04-20:
@a-bezrukov Can you have a look, please? You are listed as maintainer for Firo.

prusnak, 2021-04-20:
From the log:

```
trezorlib.exceptions.TrezorFailure: DataError: Expected input with amount
```

The client now needs to provide amount for all inputs. Please open issue for ZnodeMasternodeTool.

closed 2021-04-20 by prusnak