trezor/trezor-firmware #1577: Jump to bootloader trezorctl command

tsusanka, 2021-04-22:
Now that we have https://github.com/trezor/trezor-firmware/issues/1047 we have the ability to go to bootloader using [Protobuf message](https://github.com/trezor/trezor-firmware/blob/master/common/protob/messages-management.proto#L414). For testing purposes QA would really appreciate an easy command that would do this.

tsusanka, 2021-04-22:
`trezorctl debug send-bytes 87 ""`should do the trick!

tsusanka, 2021-04-22:
We can use that for QA purposes I believe that is enough.

closed 2021-04-22 by tsusanka

tsusanka, 2021-04-23:
Note that my instructions were incomplete (firmware sends a ButtonRequest after the aforementioned command which we need to confirm). The full tutorial is:

- Send `trezorctl debug send-bytes 87 ""`
- Do not confirm on Trezor and send `trezorctl debug send-bytes 27 ""`
- Confirm on Trezor
- Device should reboot

prusnak, 2021-04-23:
> * Send `trezorctl debug send-bytes 87 ""`
> * Do not confirm on Trezor and send `trezorctl debug send-bytes 27 ""`

You can use message names too:

* Send `trezorctl debug send-bytes RebootToBootloader ""`
* Do not confirm on Trezor and send `trezorctl debug send-bytes ButtonAck ""`