trezor/trezor-firmware #1578: Implement intermediary FW to trezorctl

tsusanka, 2021-04-22:
Now that we have https://github.com/trezor/trezor-firmware/issues/4 we can update old firmwares (such as 1.6.3) easily. We should add this intermediary FW to `trezorctl firmware-update` command. This would help a lot to our QA.

Note that @prusnak needs to upload the intermediary firmware to webwallet-data first.

tsusanka, 2021-04-22:
The intermediary FW updates 1.6.3 to the latest bootloader. So the workflow should go as:

If < 1.8.0:
- upload intermediary FW (it updates both bootloader and firmware to 1.10.0 - right @prusnak?)
- then upload any newer firmware (we should take advantage of #1047 here)

If > 1.8.0
- upload any newer firmware

Giszmo, 2021-08-23:
Not sure if I'm stuck due to this open issue or if I should file a new one.
Trezor Suite tells me to update my trezor one firmware from 1.9.4 to 1.10.2 with the device showing bootloader 1.8.0. Update fails.

tsusanka, 2021-08-23:
@Giszmo please contact support at support@trezor.io, they'll be able to troubleshoot your problem.

matejcik, 2021-10-07:
New devices ship with bootloader 1.8.0 so this is only relevant for a very small (and shrinking) user base. These users are also well served by the already existing two-update functionality.

closed 2021-10-07 by matejcik