trezor/trezor-firmware #1579: Stellar v1 TX support

alexandrubujor, 2021-04-22:
**Is your feature request related to a problem? Please describe.**
Based on personal experiments it seems that my Trezor can only sign v0 Stellar transactions and doesn't handle well the v1. I tried the latest trezor 0.12.2 python module

**Describe the solution you'd like**
Add support for v1 Stellar transactions.

**Describe alternatives you've considered**
I don't see an alternative since v0 will be deprecated sooner or later.

**Additional context**
This is the stacktrace I get when I attempt to sign a v1 TX:

Traceback (most recent call last):
  File "/Users/alexandrubujor/PycharmProjects/stellar-cli/stellar-cli.py", line 114, in <module>
    operations.send_payment(wallet_file=wallet_file, asset=asset, issuer=issuer, amount=amount,
  File "/Users/alexandrubujor/PycharmProjects/stellar-cli/stellarops/operations.py", line 201, in send_payment
    transaction = sign_trezor_transaction(transaction, k,
  File "/Users/alexandrubujor/PycharmProjects/stellar-cli/stellarops/operations.py", line 245, in sign_trezor_transaction
    tx, operations = trezor_stellar.parse_transaction_bytes(base64.b64decode(xdr_data))
  File "/Users/alexandrubujor/PycharmProjects/p39-venv/lib/python3.9/site-packages/trezorlib/stellar.py", line 89, in parse_transaction_bytes
    tx.source_account = _xdr_read_address(unpacker)
  File "/Users/alexandrubujor/PycharmProjects/p39-venv/lib/python3.9/site-packages/trezorlib/stellar.py", line 316, in _xdr_read_address
    raise ValueError("Unsupported address type")
ValueError: Unsupported address type

matejcik, 2021-07-13:
thanks
Trezor actually supports and generates V1 transactions just fine.

presumably the OP was using an unsupported address type (muxed address maybe?) in which case this is a duplicate of #1698

closed 2021-07-13 by matejcik

itsmecurtis, 2021-07-13:
I believe this issue is separate and still outstanding.

I get the same error as the original poster when trying to sign any v1 transaction, even one that has a non-muxed address (`keyTypeEd25519`, I've actually only ever tried to sign non-muxed addresses).  The error is seemingly raised before communicating with the Trezor as nothing is displayed on the screen and I'm not prompted for a passphrase.

```sh
~ $ trezorctl --version
trezorctl, version 0.12.2
~ $ trezorctl stellar sign-transaction --network-passphrase 'Test SDF Network ; September 2015' AAAAAgAAAADUj2JH1aQKLRUAe01xoqL2fsrj6cpBxml7xdDmfBadxQAAAGQABuNHAAAAAQAAAAEAAAAAAAAAAAAAAABg7fSAAAAAAAAAAAEAAAAAAAAACgAAAAt0cmV6b3JfdGVzdAAAAAABAAAABXZhbHVlAAAAAAAAAAAAAAA=
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/bin/trezorctl", line 8, in <module>
    sys.exit(cli())
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/trezorlib/cli/__init__.py", line 78, in trezorctl_command_with_client
    return func(client, *args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/trezorlib/cli/stellar.py", line 72, in sign_transaction
    tx, operations = stellar.parse_transaction_bytes(base64.b64decode(b64envelope))
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/trezorlib/stellar.py", line 89, in parse_transaction_bytes
    tx.source_account = _xdr_read_address(unpacker)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/trezorlib/stellar.py", line 316, in _xdr_read_address
    raise ValueError("Unsupported address type")
ValueError: Unsupported address type
```

alexandrubujor, 2021-07-14:
I definitely didn't try to sign a transaction with MUXED addresses as well. The addresses I use are starting with G. Please take a look at this problem and re-open it.

matejcik, 2021-07-14:
well this is interesting, because if I [strip out the envelope](https://laboratory.stellar.org/#xdr-viewer?input=AAAAANSPYkfVpAotFQB7TXGiovZ%2ByuPpykHGaXvF0OZ8Fp3FAAAAZAAG40cAAAABAAAAAQAAAAAAAAAAAAAAAGDt9IAAAAAAAAAAAQAAAAAAAAAKAAAAC3RyZXpvcl90ZXN0AAAAAAEAAAAFdmFsdWUAAAAAAAAA&type=Transaction&network=test), the V1 transaction decodes (and signs on Trezor) just fine.

There's clearly a bug in the docstring of `trezorctl stellar sign-transaction` that specifies it wants a "b64envel", but I'm wondering how this ever worked -- because that would have to mean that we supported V1 transactions wrapped in a V0 envelope? Was that something that was ever used in the wild?

matejcik, 2021-08-09:
this should now be decisively resolved by #1745