trezor/trezor-firmware #1597: Intermediary firmware deletes seed

tsusanka, 2021-05-05:
**Describe the bug**
Intermediary FW works fine but we completely forgot we also want to give it to to users that have their device **initialized**.

**Firmware version and revision**
Both intermediary 1.10.0 and 1.8.0.

**To Reproduce**
Steps to reproduce the behavior:
1. Get device with old bootloader
2. Initialize
3. Upload Intermediary firmware

**Expected behavior**
Seed is kept. The Intermediary firmware wipes it which is not what we want.

prusnak, 2021-05-05:
The fact we erase the old firmware and storage is not an oversight, but rather a feature - https://github.com/trezor/trezor-firmware/blob/625758495134698baaa9db3db571effc85e0f5ab/legacy/intermediate_fw/trezor.c#L95

However, this does not fit our user story when we'd like to use interim firmware also for people with initialized devices.

From my POV the only thing we need to change is to remove the call to `erase_firmware_and_storage` and the function itself.

This however opens a possibility for an attacker to user interim FW to downgrade to possibly vulnerable bootloader version it is shipping. This can be circumvented by calling `check_known_bootloader` and bailing out if we hit an unknown bootloader: https://github.com/trezor/trezor-firmware/blob/625758495134698baaa9db3db571effc85e0f5ab/legacy/firmware/bl_check.c#L31:L31

prusnak, 2021-05-05:
Possible fix in https://github.com/trezor/trezor-firmware/pull/1600

tsusanka, 2021-05-06:
It also [removes](https://github.com/trezor/trezor-firmware/pull/1600/files#diff-53fcfec88c0df2bbb5b860811d8bb1031491504ebc797f8bcaa1c306ab671c62L116-L119) this scary screen right?

[image]

andrewkozlik, 2021-05-10:
As I understand it, there was a valid reason why the interim firmware deleted itself and the storage and that was to remove the need for user confirmation, see https://github.com/trezor/trezor-firmware/issues/4#issuecomment-639391628. Perhaps we could check whether the storage is initialized and delete the firmware and storage only if it's not. The check doesn't need to be complicated, we just need to look at the first word of the norcow sectors. On the other hand this behavior could create some complexity for Suite. Not sure.

tsusanka, 2021-05-10:
We have agreed that we will check the storage and:
- If the storage is initialized do nothing.
- If the storage is not initialized erase both firmware and storage.

prusnak, 2021-05-13:
Closed via https://github.com/trezor/trezor-firmware/pull/1600

closed 2021-05-13 by prusnak