trezor/trezor-firmware #1598: Add exception safety to Rust code

matejcik, 2021-05-05:
Currently Rust bindings calling Micropython functions do not install exception handlers. An exception raised by Micropython will skip and throw away the Rust context, which might lead to incorrectness in certain cases.

matejcik, 2021-05-11:
Reading through https://github.com/rust-lang/rfcs/issues/2625 and https://github.com/rust-lang/rust/issues/58314, I'd be wary of using `nlr_push` in Rust directly.

For that matter, it prooobably means it's not safe to compile micropython with LLVM? Because `nlr_push` is not marked `returns_twice`. GCC also recognizes that attribute but per https://gcc.gnu.org/onlinedocs/gcc/Common-Function-Attributes.html#returns_twice it doesn't do anything problematic as long as `nlr_push` is written correctly.

This means that we should trampoline through C, per @jpochyla's suggestion. That would entail either writing wrappers for all functions that can raise, or some macro magic, or maybe some sort of universal wrapper that takes a function pointer and variadic arguments??? Either way, it's not very nice.

closed 2021-09-21 by mmilata

bosomt, 2021-09-22:
@matejcik can we move it to approved section ?

matejcik, 2021-09-22:
done

PR #1811 (2021-09-21): implement exception handling for Rust
fixes #1598