trezor/trezor-firmware #1601: Cardano streamed transaction signing

leanil, 2021-05-06:
**Is your feature request related to a problem? Please describe.**
The unserialized python representation of complete Cardano transactions can already exceed Trezor’s memory if they contain too many items (inputs, outputs, asset groups, etc.). Also new elements are continuously being added to transactions.
Independently, returning completely valid serialized transactions from Trezor won't be possible as we introduce multisig, making a rework of outgoing messages inevitable.

**Describe the solution you'd like**
The Vacuumlabs team is preparing a larger refactoring of the Cardano transaction signing process to stream the upload of transaction data to Trezor (i.e. multiple small incoming messages rather than a single `CardanoSignTx` request). We've described the proposed changes in more detail in [this document](https://docs.google.com/document/d/1K5Q8XCHt9Y2vgDNGAZed1aOJ7HokSWENT95wWBq_bJ4/edit?usp=sharing), and we would like to hear your opinion and receive feedback on it before we start implementing them.

**Describe alternatives you've considered**
No alternatives

**Additional context**
- We won’t be returning the whole serialized transaction anymore. We will only return the transaction hash, the witnesses and auxiliary data supplement. This also makes more sense for multisig, but is a breaking change for both Trezor Connect and Trezorlib.
-  The full auxiliary data as a blob will not be sent to Trezor anymore, only its hash will be - our reasoning is explained in the document.
- Could you briefly describe the intents behind Bitcoin’s implementation? E.g. what's the benefit of explicitly requesting transaction parts, rather than simply sending the parts in a predefined order?

matejcik, 2021-05-06:
> * Could you briefly describe the intents behind Bitcoin’s implementation? E.g. what's the benefit of explicitly requesting transaction parts, rather than simply sending the parts in a predefined order?

This lets the host implement a conceptually simple algorithm (send the part when requested), while the firmware can contain more complex logic of requesting certain parts of the transaction at the appropriate times. It also allows greater flexibility in processing on the firmware side.

Examples:
- Legacy signing needs to restream the whole transaction, while segwit signing only needs to restream the outputs.
- We were able to move prevtx streaming after the user confirmation, by simply requesting the outputs in advance and then requesting the previous txes afterwards -- with no change to the host-side libraries

matejcik, 2021-05-06:
> we don’t see the benefit of explicitly requesting the transaction parts (like the btc signing does)
> or even introducing empty ack messages

Empty ack (well "request" in our naming convention) messages are _required_ for the flow to actually work with client-side libraries. There is no support for just stuffing data over USB and letting the buffers keep the overflow at bay somehow :) An empty message serves the purpose of letting the host know "i'm done with this chunk, send the next one".

Regarding requesting transaction parts, I outlined some reasoning in Bitcoin above. These don't seem to apply to Cardano, where the signing flow is much saner. However, you are still relying on the host implementing the same order as the firmware and staying in sync. This assumption might or might not be reasonable, depending on the actual complexity of the flow.

matejcik, 2021-05-06:
I read over the document and the proposal seems reasonable overall. I left some comments inline.

leanil, 2021-05-06:
Thank you for the quick response, and for the clarification regarding the messaging.

tsusanka, 2021-05-10:
cc @mmahut JFYI

matejcik, 2021-09-07:
resolved by #1683

closed 2021-09-07 by matejcik