trezor/trezor-firmware #1604: Support EIP2718 EIP1559 transactions

ligi, 2021-05-08:
In the upcoming Ethereum hardfork a new style of transactions is introduced via EIP1559. It leverages the new Typed Transaction Envelope (EIP2718).
It would be great if TREZOR would support this new style of transactions.

brad-decker, 2021-05-26:
Also need this on the MetaMask team to support users who use Trezor.

matejcik, 2021-06-04:
this becomes relevant as of the London hardfork, planned for July 2021

prusnak, 2021-06-04:
> this becomes relevant as of the London hardfork, planned for July 2021

Is the old transaction format still suported after the hardfork?

prusnak, 2021-06-04:
> TREZOR users would overpay for transaction fees.

How much percent overpay are we talking here for simple transfer and simple ERC20 transfer?

brad-decker, 2021-06-04:
@prusnak the way that backward compatibility works is that gasPrice is used as the maxFeePerGas and the lack of a maxPriorityFeePerGas means that if the transaction specifies gasPrice of 100 gwei and the baseFee for the block is only 48 gwei then 52 gwei per gas is paid to the miner. Under EIP1559 the fullness of
The blocks determines both the baseFee of the next block and the amount necessary for a tip to be included in a block.

So the answer to your question is: it varies by the network conditions.

prusnak, 2021-06-04:
> So the answer to your question is: it varies by the network conditions.

That's really not helpful in deciding whether it's worth implementing this or not.

Anyways, the Ethereum community is welcome to contribute a patch via a pull request. We can't afford to spend our resources to development of this feature, but we'll gladly review the change.

timbeiko, 2021-06-04:
If anyone _is_ interested in adding this to Trezor and funds are the blocker in doing so, please ping me (tim@ethereum.org) and I can help set up a bounty for it.

FrederikBolding, 2021-06-04:
I'm gonna take a look at the code to see how difficult it would be to implement this. Will report back with any progress.

FrederikBolding, 2021-06-06:
After a bit of back and forth over the weekend (and some long nights 😴) it seems that I have a working version of the Trezor firmware that supports EIP 1559. I have opened a pull request (https://github.com/trezor/trezor-firmware/pull/1653) and we will see what the Trezor reviewers say!

angyts, 2021-08-07:
> That's really not helpful in deciding whether it's worth implementing this or not.
>
> Anyways, the Ethereum community is welcome to contribute a patch via a pull request. We can't afford to spend our [...]

`That's really not helpful in deciding whether it's worth implementing this or not.`

Wow, this line pretty much sums up my relationship with my previous favourite hardware wallet company. Not buying another trezor ever again sorry.

I mean, you guys live under a rock or something??? This EIP1559 is like the greatest event on blockchain history next to Bitcoin genesis block.

closed 2021-08-10 by matejcik

scooper4711, 2021-10-01:
This is closed - does this mean it works in both Trezor products you actively sell? Or is the Model 1 considered abandoned and won't be updated?

monomesa, 2021-10-04:
Waiting for the trezor one. :(

prusnak, 2021-10-06:
@scooper4711 @monomesa we are investigating EIP-1559 for Trezor Model One in https://github.com/trezor/trezor-firmware/issues/1834

guifel, 2021-10-14:
EIP1559 went through more than 2 months ago. I am still not able to make type 2 transactions on my Trezor one with Metamask... It's great to have to overpay and struggle to get a transaction though while all other major Hardware wallet support it.

PR #1653 (2021-08-10): Add support for Ethereum EIP 1559
Fixes #1604 

The following PR adds support for typed Ethereum transactions (EIP 2718) with a focus on EIP 1559. This changes the format of transactions to include the transaction type, new gas parameters as well as an optional access list.

It also adds a new UI screen to show the new gas parameters.
[image]

I have attempted to re-use as much of the existing transaction signing code as well as added support in the Python lib.

I have repurposed some of the existing tests to cover most of the same cases with the new format. On top of that the PR can be (and h [... cut]