trezor/trezor-firmware #1619: Device not detected by computer after upgrade to latest 1.10.0 Intermediary firmware

bosomt, 2021-05-19:
**Describe the bug**
When you update to latest provided Intermediary FW device display will blink forever after reconnection and won't be recognized by computer till you switch it to bootloader version. 
Works fine after you downgrade bootloader to older Intermediary firmware 1.10.0

**Firmware version and revision**
1.10.0 - intermediary
Firmware fingerprint: 9c97590fd86b3c9bdcaa7421f58344ba9524fa207a2673dd22ca902449401c2d

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: trezorctl
 - OS: macOS

**To Reproduce**
Steps to reproduce the behavior:
1. Upgrade to new intermediary firmware
2. Reconnect device
3. Observe device
4. Try to load full firmware to device

**Screenshots**

```
trezorctl firmware_update -f trezor-inter-1.10.0.bin
WebUSB transport is disabled: dlopen(libusb-1.0.dylib, 6): image not found
Trezor One firmware with embedded v2 image (1.8.0 or later)
Firmware version 1.10.0 build 0
Signatures are valid.
Firmware fingerprint: 9c97590fd86b3c9bdcaa7421f58344ba9524fa207a2673dd22ca902449401c2d
Embedded v2 image fingerprint: e0bd7ce4643f9c98c376a827022a001cbac8894ec7c88593525e84e19d7b69d4
Extracting embedded firmware image.
Please confirm the action on your Trezor device
testing@testings-MacBook-Air Downloads % trezorctl get_features                              
WebUSB transport is disabled: dlopen(libusb-1.0.dylib, 6): image not found
Failed to find a Trezor device.

```
https://user-images.githubusercontent.com/31506317/118867022-ec35ca80-b8e2-11eb-94b7-c011940f7c67.mp4

bosomt, 2021-05-19:
same behavior for 1.9.4/1.8.0  device /blinking + not found/

```
trezorctl get_features
WebUSB transport is disabled: dlopen(libusb-1.0.dylib, 6): image not found
Features (166 bytes) {
    bootloader_hash: 32 bytes 0xf7fa165be6d780f3e1af00abc07df8b3076bcdad72d70da22a63d8896b6391d8,
    capabilities: [
        Bitcoin (1),
        Bitcoin_like (2),
        Crypto (5),
        Ethereum (7),
        Lisk (8),
        NEM (10),
        Stellar (12),
        U2F (14),
    ],
    device_id: '6ECA4038212146D7C4976802',
    initialized: True,
    label: 'My Trezor',
    language: 'en-US',
    major_version: 1,
    minor_version: 9,
    model: '1',
    needs_backup: False,
    no_backup: False,
    passphrase_protection: True,
    patch_version: 4,
    pin_protection: False,
    revision: 20 bytes 0xffa96205fb5e22b43e7b08a3dbc3cdeee0931de3,
    unfinished_backup: False,
    unlocked: True,
    vendor: 'trezor.io',
    wipe_code_protection: False,
}
testing@testings-MacBook-Air Downloads % trezorctl firmware_update -f trezor-inter-1.10.0.bin
WebUSB transport is disabled: dlopen(libusb-1.0.dylib, 6): image not found
Please switch your device to bootloader mode.
testing@testings-MacBook-Air Downloads % trezorctl firmware_update -f trezor-inter-1.10.0.bin
WebUSB transport is disabled: dlopen(libusb-1.0.dylib, 6): image not found
Trezor One firmware with embedded v2 image (1.8.0 or later)
Firmware version 1.10.0 build 0
Signatures are valid.
Firmware fingerprint: 9c97590fd86b3c9bdcaa7421f58344ba9524fa207a2673dd22ca902449401c2d
Embedded v2 image fingerprint: e0bd7ce4643f9c98c376a827022a001cbac8894ec7c88593525e84e19d7b69d4
Extracting embedded firmware image.
Please confirm the action on your Trezor device
testing@testings-MacBook-Air Downloads % trezorctl get_features                              
WebUSB transport is disabled: dlopen(libusb-1.0.dylib, 6): image not found
Failed to find a Trezor device.
```

tsusanka, 2021-05-19:
Can you reproduce also if the device is not initialized at the beginning?

bosomt, 2021-05-19:
1) wiped device with 1.10.0 via `trezorctl wipe-device -b`
2) installed new intermediary FW
3) reconnected device
4) device not found + dark display
5) second reconnect - device detected
6) installed 1.10.0 fw via suite , works correctly

andrewkozlik, 2021-05-20:
@prusnak it looks like we should be rebooting _to the bootloader_ here: https://github.com/trezor/trezor-firmware/blob/155fc7ac67f585e04a7f3e9f1f1625aa084fab30/legacy/intermediate_fw/trezor.c#L140-L142 That's what's probably causing the blinking behavior, because the device is constantly rebooting the firmware.

Not sure about the issue with the dark display. At first glance it looks like erasing the firmware breaks the device rebooting capability, but I thought that's why we are using the `section(".data")` attribute to avoid that from happening. Now that I think about it, how does that actually work? The `.data` section gets copied from the flash to SRAM before `main()` is called, but how do we know that when we call `erase_firmware_and_reboot()` from `main()`, that it calls the copy of the function in the SRAM and not the flash. I wouldn't be surprised if the call had to be more complex to do that correctly, but I don't know enough about this to say for sure. Anyway, if this is the problem, then the simplest way to get around it seems to be not to erase the first firmware sector entirely, but to simply set the first word of the firmware header to `0x00000000`.

Where can I get trezor-inter-1.10.0.bin? It's not in https://github.com/trezor/webwallet-data/tree/master/firmware/1.

tsusanka, 2021-05-20:
> Where can I get trezor-inter-1.10.0.bin? It's not in https://github.com/trezor/webwallet-data/tree/master/firmware/1.

Sent via DM.

prusnak, 2021-05-20:
> @prusnak it looks like we should be rebooting _to the bootloader_ here:

Ah, right :-/

andrewkozlik, 2021-05-20:
So I reproduced the issue to see for myself and all the symptoms seem to be in accordance with my explanation, which doesn't mean that the explanation is right, just that it's not clearly wrong. I can't test the fix, because since the firmware is unsigned it's running into a hard fault.

andrewkozlik, 2021-05-20:
Back to the issue with the dark display which occurs when the storage is wiped (like by the bootloader for example). This is actually the main path that we should be concerned about, because it's what users with a new device are facing. The explanation I gave above is probably not the cause of the issue, because it would also appear on the first version of the 1.10.0 intermediate firmware, i.e. the one before https://github.com/trezor/trezor-firmware/pull/1600. When I tested the first version, it did not manifest the same bug, meaning that the `section(".data")` attribute seems to be working as intended. However, I can't think of any other cause at the moment. The firmware gets wiped, so the black screen clearly occurs after the wipe, but somehow the device fails to reboot.

closed 2021-05-21 by prusnak