trezor/trezor-firmware #1627: Support safety_checks parameter in ApplySettings on T1

andrewkozlik, 2021-05-21:
In 568da98199 which was released with T1 firmware v1.9.2 we introduced a safety check that doesn't allow spending LTC from a Bitcoin path, because LTC doesn't have strong replay protection via SIGHASH_FORKID. In order to claim coins sent to a wrong path users had to downgrade to firmware 1.9.1 or 1.9.0 and then use Electrum to recover the coins. Now that we have a new bootloader with firmware 1.10.0, it is no longer possible to downgrade to earlier firmware versions, i.e. older firmware will not run at all. So users have no way of claiming these coins other than buying a TT and entering the seed there or entering it directly into Electrum, which are both pretty harsh options. The solution is to allow users to disable safety checks in T1 like they can on TT.

prusnak, 2021-05-21:
Quick fix would be replacing the following

https://github.com/trezor/trezor-firmware/blob/4f1274c9c4eb3c74feb8f9605bace2285b3199c3/legacy/firmware/signing.c#L687-L690

with the code taken from

https://github.com/trezor/trezor-firmware/blob/4f1274c9c4eb3c74feb8f9605bace2285b3199c3/legacy/firmware/fsm_msg_coin.h#L186-L196

but adapted to `return false;` instead of `return;`

Not sure if we even want/need to add the whole `safety_checks` thing. But surely we can if we want ...

andrewkozlik, 2021-05-21:
> Quick fix would be [...]

Yes, that's a very good point that we can just behave in T1 as if safety_checks == PromptAlways. But I think I already have it implemented:

[image] [image]

I could also implement the `PromptAlways` setting fairly quickly, but I am not sure whether it's desired.

tsusanka, 2021-05-21:
We should also not forget reporting the safety_checks value in Features. Suite depends on it (see my comment [here](https://github.com/trezor/trezor-suite/issues/3405#issuecomment-846100197)).

prusnak, 2021-05-21:
> I could also implement the `PromptAlways` setting fairly quickly, but I am not sure whether it's desired.

I am torn on this one. On one hand we don't even allow setting PromptAlways via Suite, so I am inclined to leave that one out; but OTOH I like things to be consistent :-) I leave that one for you to decide, @tsusanka

tsusanka, 2021-05-21:
I just wanted to write something very similar :). So I think it is up to you @andrewkozlik, if it is an easy thing let's do it, but let's not worry about it if it is not.

andrewkozlik, 2021-05-21:
Added a new test `test_invalid_path_prompt` in test_msg_signtx_invalid_path.py:

[image] [image] [image]

andrewkozlik, 2021-05-24:
Test case for @qa:
Send some LTC to a BTC address on T1 and then try to reclaim the amount by sending it back to a LTC address using Electrum.
1. With safety checks set to "strict" the T1 should decline to sign the transaction.
2. With safety checks set to "prompt" the T1 should show a warning before signing the transaction.

Use Trezor Suite to disable/enable the safety checks. Note that after unplugging Trezor the setting reverts to "strict".

closed 2021-05-24 by andrewkozlik

bosomt, 2021-05-24:
Does suite need any kind of adjustment to support this feature ?
Tried twice with new version of firmware /1.10.1 commit d80d5456 / , got error /see screenshot/ and device froze both times.

results from trezorctl

```
 trezorctl set safety-checks prompt
Error: ProcessError: Unsupported safety-checks setting

trezorctl version
0.12.2

```

[image]

**Info:**
 - Suite version: web 21.6.0 (a1352378b29267f215fa5658d366af9923a2aae9)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/90.0.4430.212 Safari/537.36
 - OS: MacIntel
 - Screen: 1440x900
 - Device: trezor-fw-regular-1.10.1-d80d5456.bin

andrewkozlik, 2021-05-24:
1. The problem with `trezorctl` is that version 0.12.1 sends `PromptAlways`, but I was testing with the version in the repo, which sends `PromptTemporarily`. (I didn't implement `PromptAlways`, because it's not in Suite and I thought it wasn't in `trezorctl` either plus there was opposition to this setting from the product team to this in the past.) So the version of `trezorctl` in the repo works fine. I don't know if there is an option to make the released version send `PromptTemporarily` :-(.
2. Works for me with Trezor-Suite-21.5.1-linux-x86_64 and with the current released web version. Unfortunately I can't get the development version of Suite for web at `suite.corp.sldev.cz` running, because it's claiming it can't find the bridge, so not sure what the problem is there.

andrewkozlik, 2021-05-24:
The unreleased desktop version of Suite (Trezor-Suite-21.6.0-linux-x86_64) also works for me, so I am not sure what's wrong with the web version.

The released version of `trezorctl` apparently doesn't have a way to send `PromptTemporarily`, so @matejcik will need to release a new version.

bosomt, 2021-05-25:
1) yes bridge doesn't work at sldev domain due to CORS protection :( :( so we have to use webusb , maybe this is causing problems because other versions of suite doesn't have problem to change safety checks settings. 

2) current app master works as expected

**Info:**
 - Suite version: desktop 21.6.0 (a1352378b29267f215fa5658d366af9923a2aae9)
 - OS: MacIntel
 - Device: model One 1.10.1 regular

3) current web staging works as expected in Chrome and in Firefox

**Info:**
 - Suite version: web 21.5.1 (4d09d88476dab2e6b2fbfb833b749e9ac62251c2)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:88.0) Gecko/20100101 Firefox/88.0
 - OS: MacIntel
 - Screen: 1440x900
 - Device: model One 1.10.1 regular

**Info:**
 - Suite version: web 21.5.1 (4d09d88476dab2e6b2fbfb833b749e9ac62251c2)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/90.0.4430.212 Safari/537.36
 - OS: MacIntel
 - Screen: 1440x900
 - Device: model One 1.10.1 regular

My1, 2021-09-12:
does this also work for other coin types like Ethereum and ethereum classic (a pair where I had seen multiple times that people had sent accidentially)