trezor/trezor-firmware #1630: pb2py error

Momo10101, 2021-05-25:
Hello,

I am trying to convert protobuf .proto definitions to python objects using pb2py, however, i get the following error:

> ./pb2py messages-tezos.proto 
> WARNING:root:Message IDs not found under 'MessageType'
> Traceback (most recent call last):
>   File "./pb2py", line 481, in <module>
>     descriptor.write_classes(tmpdir, not args.no_init_py, args.include_deprecated)
>   File "./pb2py", line 454, in write_classes
>     self.process_messages(self.messages, include_deprecated)
>   File "./pb2py", line 422, in process_messages
>     self.write_to_file(
>   File "./pb2py", line 437, in write_to_file
>     for line in out:
>   File "./pb2py", line 344, in process_message
>     msg_id = self._get_extension(message, "wire_type")
>   File "./pb2py", line 223, in _get_extension
>     extension_num = self.extensions[extension_name]
> KeyError: 'wire_type'

I am doing this on a clean clone of the trezor firmware on an Ubuntu 20.04. Unfortunately pb2py is not documented in the developer guide so I am not sure if I am doing something wrong.

matejcik, 2021-05-25:
please use pb2py through the `make protobuf` make target

closed 2021-05-25 by matejcik

Momo10101, 2021-05-25:
Ok thanks, will do, but it would be nice to know how to use pb2py directly as well.

prusnak, 2021-05-25:
> it would be nice to know how to use pb2py directly as well.

You can always read the source code of the script that is being called via `make protobuf`: https://github.com/trezor/trezor-firmware/blob/master/tools/build_protobuf

Momo10101, 2021-05-25:
Perfect, thanks a lot!

Momo10101, 2021-05-26:
I have a couple of more questions i hope you can help me with. I assume that if I am adding a new coin I should modify `build_protobuf` and add `messages-mycoin.proto` to `CORE_PROTOBUF_SOURCES`. Also i should add my messages to `/common/protob/messages.proto`, for example: `MessageType_MyCoinGetPublicKey = 89 [(wire_in) = true];` under `enum MessageType`. 

Now i am not sure if this is a bug, but when i do that, my messages don't get indented properly in file  `src/trezor/messages/MessageType.py`. I get this:

```
if not utils.BITCOIN_ONLY:
    EthereumGetPublicKey: Literal[450] = 450
    EthereumPublicKey: Literal[451] = 451
    EthereumGetAddress: Literal[56] = 56
    EthereumAddress: Literal[57] = 57
    EthereumSignTx: Literal[58] = 58
    EthereumTxRequest: Literal[59] = 59
    EthereumTxAck: Literal[60] = 60
    EthereumSignMessage: Literal[64] = 64
    EthereumVerifyMessage: Literal[65] = 65
    EthereumMessageSignature: Literal[66] = 66
  ...
MyCoinGetPublicKey: Literal[89] = 89
  ...

```
I have to manually indent it myself, and i dont understand why?

Also the python .py files that are autogenerated from `messages-mycoin.proto` don't contain `MESSAGE_WIRE_TYPE = 89` as a class field. I checked the other .proto files and my .proto file looks very similar so I dont understand why I am not getting that field. Could you tell me how?

matejcik, 2021-05-27:
1) the indentation is now done explicitly in `build_protobuf` so you'd need to edit that

2) you should get MESSAGE_WIRE_TYPE if for `MyCoinSignTx` there is an entry `MessageType_MyCoinSignTx` in messages.proto, AND that message has an unique id. (i'm not sure if protoc even lets you compile if ids are non-unique)

3) are you aware that we are currently not accepting new coins into the firmware? just making sure

Momo10101, 2021-05-27:
Thanks for the quick reply @matejcik: 

1. Ok i got it working now, thanks! A part from from adding the coin to `CORE_PROTOBUF_SOURCES` in `build_protobuf`, it also needs to be added to the `altcoin` "for loop", otherwise it wont be indented.

2. Perfect, i have it working now as well

3. Yes, i am aware. Just playing around for now : )

Do you guys have a slack or discord channel or something for developers?

matejcik, 2021-05-27:
there's gitter https://gitter.im/trezor/community
it's not great ;)