trezor/trezor-firmware #1639: Update QR-code-generator

matejcik, 2021-06-01:
https://github.com/nayuki/QR-Code-generator

couldn't find a changelog, reviewing all commits shouldn't be too difficult though

I also notice they have a Rust version! and it's available as a crate too. Maybe investigate switching to that?

nayuki, 2021-06-03:
I see that the [current version of this Trezor repository](https://github.com/trezor/trezor-firmware/tree/767e7b8e8de3cd677211406d52c68a3da10915b7/vendor) has pinned my [QR lib at a version from 2019-02-14](https://github.com/nayuki/QR-Code-generator/tree/40d24f38aa0a8180b271b6c88be8633f842ed9d4).

Trezor is using the C language version of my library. What has [changed](https://github.com/nayuki/QR-Code-generator/commits/8518684c0f33d004fa93971be2c6a8eca3167d1e/c) between then and [now](https://github.com/nayuki/QR-Code-generator/tree/8518684c0f33d004fa93971be2c6a8eca3167d1e) in the C port:

* Slightly different algorithm for detecting finder patterns for penalty calculation.
* Improved the correctness of handling integers - types, bit widths, signedness, arithmetic, etc.
* Revamp some private/internal functions and variables.

For the C lib, when moving from the 2019-02-14 version to the 2020-09-12 version, the impact on callers should be small.

nayuki, 2021-06-03:
My C library completely avoids heap allocations; it expects the caller to pass in input/output buffers and scratch space. This is not true for my other language ports; all of them do heap allocations and deallocations. So neither my C++ lib nor Rust lib match this feature of my C lib. Also, my Rust lib doesn't define a C foreign function interface (FFI) or do anything to help users coming from other languages.

prusnak, 2021-06-03:
@nayuki Thanks for the summary! It seems like it makes sense to stick to the C version for now.

Maybe in the future when we rewrite the whole rendering stack to Rust and when qrcodegen crate has a `#![no_std]` option with no allocations, it would make sense to switch.

closed 2021-11-05 by mmilata

bosomt, 2021-11-16:
@mmilata any special testing needed on our side or just test/validate/compare QR code generated by T1 and TT in current dev master ?

nayuki, 2021-11-16:
> Maybe in the future [...] when qrcodegen crate works with `#![no_std]` option and has no allocations

FYI I made this a reality last week: https://github.com/nayuki/QR-Code-generator/tree/master/rust-no-heap

mmilata, 2021-11-16:
@bosomt generating and validating a QR code on T1 and TT should be enough

@nayuki neat! we might switch to that eventually

bosomt, 2021-11-16:
**QA OK** trezor-fw-regular-2.4.3-221977ad.bin
**QA OK** trezor-fw-regular-1.10.4-221977ad.bin

BTC receive QR code OK

PR #1894 (2021-11-05): chore(vendor): update QR-code-generator
Fixes #1639. The only non-cosmetic change seems to be the improved finder pattern detection. Bumped to master since there are only variable name changes since the 1.7.0 release. Code size seems to stay the same.

~~Waiting for nayuki/QR-Code-generator#127.~~