trezor/trezor-firmware #1643: Spaces lost in Shamir info texts

bosomt, 2021-06-03:
**Describe the bug**
See screenshots. Looks like we lost all spaces in Shamir info texts.

**Firmware version and revision**
major_version: 2,
    minor_version: 4,
    model: 'T',
    needs_backup: False,
    no_backup: False,
    passphrase_always_on_device: False,
    passphrase_protection: True,
    patch_version: 0,
    pin_protection: False,
    recovery_mode: False,
    revision: 9 bytes b'3204fd682',
    sd_card_present: False,
    sd_protection: False,
    unfinished_backup: True,
    unlocked: True,
    vendor: 'trezor.io',
    wipe_code_protection: False,

**Info:**
 - Suite version: web 21.5.1 (4d09d88476dab2e6b2fbfb833b749e9ac62251c2)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:88.0) Gecko/20100101 Firefox/88.0
 - OS: MacIntel
 - Screen: 1824x1140
 - Device: model T 2.4.0 Bitcoin only

**To Reproduce**
Steps to reproduce the behavior:
1. Start Shamir backup
2. Click on INFO button during backup
3. Observe

**Screenshots**
[image]

[image]

closed 2021-06-03 by mmilata

bosomt, 2021-06-03:
**QA OK**

fixed in trezor-fw-regular-2.4.0-ea3596ad.bin

mmilata, 2021-06-03:
@bosomt thanks!

PR #1648 (2021-06-03): fix(core): spaces in Shamir info texts 
Fixes #1643.

[image][image][image][image][image]