trezor/trezor-firmware #1651: Add changelog checker to GitHub actions

matejcik, 2021-06-04:


vdovhanych, 2021-07-19:
Hi, maybe it's very straightforward, but I'm not sure what it is that should be checked here. Assume it's something like [this](https://github.com/marketplace/actions/changelog-checker). Thnx @matejcik

matejcik, 2021-07-20:
we have a gitlab job for this, and a custom checker script in `ci/check-changelog.sh`
we just want to run that in github actions (as well), mainly so that external contributors get the result separately and straight away

cc @mmilata who made the script

matejcik, 2021-07-20:
this changelog check should not block merging, we sometimes want to deliberately ignore the result

matejcik, 2021-07-23:
on #1706 the github action is passing and the gitlab job is failing. the github action should fail as well.

vdovhanych, 2021-07-23:
Interesting, the only thing I can think of is that GitHub is cloning the repository is differently from GitLab. @mmilata maybe that variable is indeed needed for the script?

mmilata, 2021-07-23:
That shouldn't be the problem, the script correctly detected that it's a feature (not release) branch. Maybe GH actions point `origin/master` to something else than GitLab does?

vdovhanych, 2021-07-23:
Could this be the problem @mmilata? 

Checkout pull request HEAD commit instead of merge commit
```yaml
- uses: actions/checkout@v2
  with:
    ref: ${{ github.event.pull_request.head.sha }}
```

Now it checks out HEAD of the merge commit would changing to this help?

mmilata, 2021-07-23:
Hmm, I tested that and the script seems to work fine on merge commits. Looking at the options I wonder if the lack of history due to default `fetch-depth: 1` isn't the problem. Let's try with `fetch-depth: 0`?

closed 2021-07-26 by mmilata