trezor/trezor-firmware #1654: Remove non-Bitcoin enums from Bitcoin-only build

matejcik, 2021-06-07:
depends on #1541 because of the significant changes there

from @mmilata's comment:
> Though some enums still make it to the image:
> 
> ```
> $ tools/check-bitcoin-only core/build/firmware/firmware.bin
> Decred details provided but Decred coin not specified.
> DecredStakingSpendType
> )C#trezor.enums.DecredStakingSpendType
> .Q&trezor/enums/DecredStakingSpendType.py
> UiConfirmDecredSSTXSubmission
> trezor/enums/DecredStakingSpendType.py
> NEMImportanceTransferMode
> &trezor.enums.NEMImportanceTransferMode
> NEMModificationType
>  trezor.enums.NEMModificationType
> NEMMosaicLevy
> trezor.enums.NEMMosaicLevy
> NEMSupplyChangeType
>  trezor.enums.NEMSupplyChangeType
> CardanoAddressParametersType
> CardanoBlockchainPointerType
> CardanoAddressType
> trezor.enums.CardanoAddressType
> CardanoCertificateType
> #trezor.enums.CardanoCertificateType
> CardanoPoolRelayType
> !trezor.enums.CardanoPoolRelayType
> LiskTransactionType
>  trezor.enums.LiskTransactionType
> StellarAssetType
> MoneroAccountPublicAddress
> MoneroMultisigKLRki
> MoneroOutputEntry
> MoneroRctKeyPublic
> !MoneroTransactionDestinationEntry
> MoneroTransactionRsigData
> MoneroTransactionSourceEntry
> TezosBallotType
> trezor.enums.TezosBallotType
> TezosContractType
> trezor.enums.TezosContractType
> ```

mmilata, 2021-06-22:
Build system & legacy: #1633

matejcik, 2021-10-16:
not much to test for QA