trezor/trezor-firmware #1656: Support Taproot

tsusanka, 2021-06-08:


prusnak, 2021-06-08:
FTR:  Taproot softfork will activate in November.

tsusanka, 2021-06-08:
@prusnak could you fill in the last bullet point please? I did catch it properly what was it.

prusnak, 2021-06-08:
> @prusnak could you fill in the last bullet point please? I did not catch it properly what was it.

Done

andrewkozlik, 2021-06-22:
Additional tasks:
- [ ] In `SignTx` implement the BIP-341 "common signature message" computation.
- [ ] In `SignTx` disable streaming of previous transactions if at least one input is taproot.
- [ ] Implement taproot support in `scripts.py`, i.e. `output_script_native_p2tr()` , `witness_p2tr()`, `parse_witness_p2tr()` etc.
- [ ] Decide how best to implement multisig for taproot in Trezor
    - MuSig (interactive signing) vs. unagreggated multisig (non-interactive).
    - Consider that taproot allows encoding more complicated schemes, e.g. with four keys A, B, C and D, we can have something like "A or (B and C) or (B and D)". This would require modifications to `MultisigRedeemScriptType`.
- [ ] Implement multisig scripts for taproot. 
- [ ] Use new Bech32m format in `GetAddress`.
- [ ] Support taproot scripts and Schnorr signatures in `SignatureVerifier`.
- [ ] Support taproot scripts and Schnorr signatures in `GetOwnershipProof` and in `verify_nonownership()`.
- [ ] Figure out how to deal with taproot in `SignMessage` and `VerifyMessage`. Bitcoin Core doesn't seem to use Schnorr signatures for signing messages. This is related to AOPP support https://github.com/trezor/trezor-firmware/issues/1586.

prusnak, 2021-06-22:
> [ ] Decide how best to implement multisig for taproot in Trezor

I prefer to transfer the multisig related tasks to a separate topic (new epic issue) and focus this effort only for single-sig Taproot.

andrewkozlik, 2021-06-23:
> - [ ] Decide BIP-44 paths

Andrew Chow is proposing to use a new purpose number for taproot paths:
https://lists.linuxfoundation.org/pipermail/bitcoin-dev/2021-June/019096.html
https://github.com/achow101/bips/blob/taproot-bip44/bip-taproot-bip44.mediawiki

bosomt, 2021-11-22:
**QA OK**

gruve-p, 2021-11-22:
Is the support for Taproot also being implemented for altcoins like GRS? GRS has activated Taproot since june 2021 and Electrum-GRS has support for sending to Taproot addresses.

matejcik, 2021-11-22:
@gruve-p not at this time, but please create a separate issue. Enabling Taproot for other coins should not be difficult.

tsusanka, 2021-11-25:
Let's close this and separate some of the subtasks from Andrew into new issues?

andrewkozlik, 2021-11-25:
> Let's close this and separate some of the subtasks from Andrew into new issues?

Done: https://github.com/trezor/trezor-firmware/issues/1943, https://github.com/trezor/trezor-firmware/issues/1944, https://github.com/trezor/trezor-firmware/issues/1946.

closed 2021-11-25 by andrewkozlik

gruve-p, 2021-12-20:
@matejcik My attempt to enable Taproot for GRS: https://github.com/trezor/trezor-firmware/pull/2030