trezor/trezor-firmware #1657: Cache monero test chain in CI

matejcik, 2021-06-09:
per @ph4r05:

> pustas to s cacheovanim blockchainu? `--chain /path/to/chain/file` Ak je to pustene bez tohoto parametru, pri kazdom spusteni to generuje novy testovaci chain, to nejaky cas trva. Subor nemusi existovat. Po prvom generovani sa to tam ulozi a dalsie pustenie to len loadne
v Travise by som to napr dal do nejakej build cache, nech je to rychlejsie. Pripadne lokalne to takto ide v pohode. 
(moze byt problem, ked sa zmeni nejak format, vtedy to treba pregenerovat. Nie som si 100% isty, ci to ta binarka vie vyriesit, alebo proste po padnutych testoch zmazat chain file)

the initial chain generation takes 60-100s for me. 

CI run should cache the chain file so that it doesn't need to be regenerated every time.

I fixed the issue for local runs in fa98c6c

---

In addition, even with chain caching, there's still a 20s pause when the daemon is presumably verifying the chain data. This verification isn't cached anywhere. It would be nice to cache that too; however, we'd need to modify the testing executable.

matejcik, 2021-07-23:
@vdovhanych assigning to you but it's low priority

ph4r05, 2021-07-27:
my view: Modification is IMO required to take place in Monero code as the verification is run by monero-daemon. I can imagine that changing daemon code has to be pretty well justified as it is a critical component of the Monero network. If the modification does not make it to the upstream as PR, I don't have capacity to maintain side modifications outside upstream. But if you will maintain this, it could work.