trezor/trezor-firmware #1658: Successful wipe code triggers RSOD

wendys-cats, 2021-06-09:
**Describe the bug**
After entering wipe code, device will successfully wipe, but will display RSOD. After reconnecting the device is wiped and functions. Applies only for TT device 2.4.0 firmware, T1 works as intended. 

Text
FATAL ERROR
expr: drbg_initialized
file: embed/trezorhal/random&delays.c:81
func: drbg_generate
rev:  revisbelow

**Firmware version and revision**
TT 2.4.0 usigned ea3596a
TT 2.4.0 signed ea3596ad8

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Trezor Suite app and staging wallet 
 - OS: Windows and NixOS
 - Browser none and Chromium 
 - Version APP and Version 88.0.4324.182 (Official Build) (64-bit)

**To Reproduce**
Steps to reproduce the behavior:
1. Enable PIN protection
2. Enable wipe-code protection via trezorctl
3. Enter wipe-code instead of PIN
4. See error

**Expected behavior**
Device will wipe itself without displaying errors.

**Screenshots**

[image]

prusnak, 2021-06-09:
This is a regression from 2.3.x, assigning P1

STew790, 2021-06-09:
I was able to reproduce this with exactly the same setups. 

**Firmware version and revision**
TT 2.4.0 usigned ea3596a
TT 2.4.0 signed ea3596ad8

**Desktop/smartphone setup (please complete the following information):**

Wallet software: Trezor Suite app and staging wallet 3a553b24e498c56a2e03897fded706dffb4a6d6b
OS: Windows 10 and NixOS 21.03
Browser none and Chromium
Version APP and Version 88.0.4324.182 (Official Build) (64-bit)

andrewkozlik, 2021-06-09:
This is happening because the `shutdown()` function defined in `core/embed/trezorhal/util.s` erases the RAM which causes the `drbg_initialized` static variable to be set to `false` while the random interrupts are running. `drbg_generate()` doesn't expect this, so it runs into a fatal error screen. The `shutdown()` function is called for all of the red screens on Trezor, so this means that any red screen will end up in the "FATAL ERROR drbg_initialized" screen shown in the screenshot. One of the red screens shown on Trezor is the one which states "You have entered the wipe code. All private data has been erased". So this is what's happening here. The Trezor gets wiped correctly but the information screen is then replaced by the FATAL ERROR screen.

onvej-sl, 2021-06-22:
What `shutdown()` does is:
  * It disables all exceptions (including interrupts) that can be disabled.
  * It erases the whole RAM and registers.
  * It ends up in an infinite loop.

Note that disabling all the exceptions is important, since an exception (for example the systick handler) could access memory that is already cleared which would lead to another exception.

The cause of this issue is that if `shutdown()` is called from the unprivileged mode, the function fails to disable all the exceptions.

closed 2021-06-23 by onvej-sl

bosomt, 2021-07-04:
**QA OK**

**Info:**
 - Suite version: desktop 21.6.2 (290aab68e418b837f9bed8a8839143dddaefb81c)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 11_4_0) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/21.6.2 Chrome/89.0.4389.69 Electron/12.0.0 Safari/537.36
 - OS: MacIntel
 - Screen: 1680x1050
 - Device: model T 2.4.1 regular 0x01c1ae426fe911619f0312fa417ecdd7e102848c

wendys-cats, 2021-07-13:
**QA OK**

No more RSOD, yay! After successful wipe, device displays message about entering wipe code and not RSOD.

**Suite**: version APP 21.6.2 290aab68e418b837f9bed8a8839143dddaefb81c **and** APP 21.7.1 3d90909aa68da250490f474b2d4cab02c0aad272
**OS**: Windows 10
**Device**: TT 2.4.1 regular 01c1ae426fe911619f0312fa417ecdd7e102848c