trezor/trezor-firmware #1659: BAD 2.4.0 Bug found

try777772, 2021-06-09:
Trezor Connect will NOT work with a blank passphrase after updating to 2.4.0

Downgrade to 2.3.6 and it works again.

For example:
When you go to adalite.io, trezor-connect asks for your passphrase like it always does.  It has a message to leave it blank for standard non-hidden wallet.

If you leave it blank on 2.4.0, it will never go past that screen.  It continues to re-ask you over and over.  Broken.

If you leave it blank on 2.3.6, it opens the wallet properly.  Works.

Windows 10, Chrome current version.

uj, 2021-06-09:
I can reproduce this as well after updating to trezor firmware 2.4.0

prusnak, 2021-06-09:
@try777772 @uj Does it happen on wallet.trezor.io or just via Connect?

prusnak, 2021-06-09:
The empty passphrase works for me on wallet.trezor.io and suite.trezor.io 

@try777772 Does it happen for you on other wallets too or just Adalite?

@uj What wallet did you use to reproduce?

uj, 2021-06-09:
I tried several online wallets, and these 2 are the only ones I found so far that do not work:
adalite.io
mycrypto.com

You don't actually need any crypto in your account to try it, you don't even need to have ever visited the site before.  Just plug in the trezor, go to the website, and connect with a blank password.

uj, 2021-06-09:
I also googled it and found a reddit thread for it

https://www.reddit.com/r/TREZOR/comments/nw5cq4/adalite_not_working_after_upgrade_240/

lunarpool, 2021-06-10:
I'm able to log in to Adalite with blank passphrase with the following sequence:

1) unplug/plug Trezor T
2) unlock with my PIN
3) start Chrome and open adalite.io
4) Choose unlock with Trezor
5) Confirm public keys Export
6) Confirm blank passphrase on Trezor T
7) Confirm blank passphrase on Trezor T again
8) Confirm passphrase with space character on Trezor T

Weird, but tried several times and works perfectly fine.

lunarpool, 2021-06-10:
Localy installed SW wallet for Cardano -  **Daedalus** works fine, **cardano-hw-cli** not affected as well

tsusanka, 2021-06-10:
@trezor/qa please retest. On both adalite and in Suite.

lunarpool, 2021-06-10:
Adalite - still not working, same behaviour, asking for passphrase in an infinite loop
Suite - works fine

bosomt, 2021-06-10:
@tsusanka i can confirm loop behaviour in adalite wallet
suite works fine, we already tested it with 2.4.0 multiple times

https://user-images.githubusercontent.com/31506317/121508020-76a2b300-c9e5-11eb-8e0d-b3447ad00155.mov

tsusanka, 2021-06-10:
@bosomt thx, could you please also try other Connect-based? E.g. mycrypto or similar

matejcik, 2021-06-10:
Confirmed in firmware.

The problem is that the new cache conflates `""` with "nothing is cached". I have a quick patch, but I am not sure if that's a good final solution.

prusnak, 2021-06-10:
> The problem is that the new cache conflates

What coins are affected?

matejcik, 2021-06-10:
User-side workaround is clicking *Enter* repeatedly until the dialog goes away. ISTM it always takes seven clicks to load, one or two for every subsequent operations. @bosomt if you're more familiar with adalite, can you confirm?

This is AFAICT the principle behind @lunarpool's method: you enter the empty passphrase enough times to fetch the appropriate data, and then use `" "` which gets cached and lets you through.

matejcik, 2021-06-10:
> > The problem is that the new cache conflates
> 
> What coins are affected?

only Cardano

bosomt, 2021-06-10:
@tsusanka  mycrypto works as expected, previous transactions/used addresses are loaded

lunarpool, 2021-06-10:
> This is AFAICT the principle behind @lunarpool's method: you enter the empty passphrase enough times to fetch the appropriate data, and then use `" "` which gets cached and lets you through.

Yup, I can confirm this. Adalite works fine for me when I confirm the blank passphrase 7 times (on the device)

sorooris, 2021-06-10:
Confirming @matejcik's and @lunarpool's method - took me ~6 confirmations of empty passphrase to get into Adalite wallet and ~4 confirmations on mycrypto.

matejcik, 2021-06-10:
> ~4 confirmations on mycrypto.

Is mycrypto also able to access Cardano? I can't reproduce there.

sorooris, 2021-06-10:
> > ~4 confirmations on mycrypto.
> 
> Is mycrypto also able to access Cardano? I can't reproduce there.

I'm so sorry, no Cardano there afaik, we were asked to test mycrypto on Slack (where there's a different behavior in which the Connect popup opens about 4 times). I'm not used to testing 3rd party apps, my mistake.

tsusanka, 2021-06-10:
Btw just a note: we have intentionally [not released](https://github.com/trezor/connect/pull/830) Connect yet. So JFYI at least Cardano-only users should not be affected. I know that is a minority but at least something.

matejcik, 2021-06-10:
@sorooris just to be clear: mycrypto opens the pop-up multiple times, but only asks for passphrase _once_ and the subsequent popups do not ask, correct?

sorooris, 2021-06-10:
@matejcik correct :arrow_down: 
Connect pop-ups:
- Export public key of Ethereum account #1
- Enter My Trezor passphrase Note: Passphrase is case-sensitive.
- Export multiple public keys
- Export multiple public keys again

closed 2021-06-11 by matejcik

tsusanka, 2021-06-11:
@trezor/qa could one of you please retest on adalite.io? Someone on reddit [reported](https://www.reddit.com/r/TREZOR/comments/nw5cq4/adalite_not_working_after_upgrade_240/h1a4p9j/?utm_source=share&utm_medium=web2x&context=3) it seems fixed so I am wondering if Adalite folks did some hotfix.

bosomt, 2021-06-11:
@tsusanka nah, still same result, adalite+connect asks for passphrase over and over again

refi93, 2021-06-11:
To clarify - Adalite exports multiple keys in bulk as it has support for multiple accounts and this bug results for the prompt being triggered for every single key. So after 5-6 confirmations (the amount of keys exported in bulk) the user will indeed eventually log in. Of course this is still a significant hurdle, unfortunately the only option is to wait for the fix on the connect side

A partial fix (in case the users use only one account in Adalite) for the time being is to disable bulk key export (there's a checkbox for it below the hardware wallet selection screen), in which case "only" three keys get exported and hence "only" three prompts are triggered. Of course, as soon as the problem in trezor connect is fixed, it's worth re-enabling it.

bosomt, 2021-07-04:
**QA OK**

**Info:**
 - Suite version: desktop 21.6.2 (290aab68e418b837f9bed8a8839143dddaefb81c)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 11_4_0) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/21.6.2 Chrome/89.0.4389.69 Electron/12.0.0 Safari/537.36
 - OS: MacIntel
 - Screen: 1680x1050
 - Device: model T 2.4.1 regular 0x01c1ae426fe911619f0312fa417ecdd7e102848c

commit 18587852 (2021-06-10): fix(core): allow caching empty values (fixes #1659)
commit 8b61f8ce (2021-06-11): fixup! fix(core): allow caching empty values (fixes #1659)
commit 7df539d3 (2021-06-11): fix(core): allow caching empty values (fixes #1659)
commit 52c34c73 (2021-06-11): fix(core): allow caching empty values (fixes #1659)