trezor/trezor-firmware #1660: Trezor model one update breaks Unchained confirm on device

waldenraines, 2021-06-09:
**Describe the bug**
The confirm on device feature no longer works with the latest version of the model one.  It seems like the exclusions from https://github.com/trezor/trezor-firmware/blob/master/core/src/apps/bitcoin/keychain.py#L56-L62 aren't being applied and anything that doesn't match https://github.com/trezor/trezor-firmware/blob/f93a8514e8e25b17b02360d9955c0a30999adf68/legacy/firmware/crypto.c#L526 fails with `Forbidden key path`

This appears to be the change that broke it:

https://github.com/trezor/trezor-firmware/commit/2dbde040c43b17b8bad9e908397b578404b402ca#diff-48ff7a04a5484acc77257bbd73e21a02a4508adb384582a4c8197d85c07270f1R188-R193

**Firmware version and revision**
`0.10.1` only, works with `0.10.0`

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: caravan, Unchained.com
 - OS: All desktops
 - Browser: all
 - Version: all

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://unchained-capital.github.io/caravan
2. Create a multisig wallet by uploading two or more xpubs
3. Expand first address on the addresses tab
4. Click confirm on device tab
5. Select your key
6. Select trezor
7. Click confirm and go through trezor connect flow
8. Note error (`Forbidden key path`)

**Expected behavior**
Confirm on device works

tsusanka, 2021-06-10:
I will follow-up via email shortly.

andrewkozlik, 2021-06-10:
**Workaround**

A workaround is for the user to disable safety checks, which they can do via Trezor Suite.

**History**

Unfortunately we still don't have device tests for the Unchained Capital paths, which would have nipped this in the bud. I already created an issue https://github.com/trezor/trezor-firmware/issues/1665.

For reference the issue that started this off was https://github.com/trezor/trezor-firmware/issues/1627, which deals with the problem of users accidentally sending LTC to BTC paths and the lack of strong replay protection in LTC.

Trezor 1 seems to be doing path checks quite differently to Trezor T and I don't feel now is the time to overhaul this on T1. Feel free to create an issue though. What T1 does is two types of path checks: full and partial.
- For `SignTx` only partial checks are done and if they didn't succeed, then signing used to fail. In https://github.com/trezor/trezor-firmware/commit/2dbde040c43b17b8bad9e908397b578404b402ca I changed this to show a warning when safety checks are disabled and fail otherwise. This resolved #1627.
- For `GetAddress` full checks are done and if they didn't succeed, then a warning used to be shown. In https://github.com/trezor/trezor-firmware/commit/2dbde040c43b17b8bad9e908397b578404b402ca I changed this to show a warning when safety checks are disabled and fail otherwise, to be consistent with the previous case. This caused the present issue.

**Possible solutions**
1. Revert the changes made in https://github.com/trezor/trezor-firmware/commit/2dbde040c43b17b8bad9e908397b578404b402ca to the `GetAddress` checks.
2. Add the non-standard paths listed [here](https://github.com/trezor/trezor-firmware/blob/74cf309a9311ee6ec33cea6f7b7bab1c5a15aedf/core/src/apps/bitcoin/keychain.py#L48-L62) to `coin_known_path_check()`.
3. Handle path checks in `GetAddress` and `SignTx` the same way and show warning for non-standard paths.
4. Get rid of safety checks setting altogether in T1 and just show warnings when partial or full path checks don't pass.

The fact that path checks in `GetAddress` and `SignTx` are handled differently seems confusing, which is why at first glance I find Option 3 with the following logic most appealing:
- If partial path checks pass but full checks don't, then show a warning.
- If partial path checks don't pass and safety checks are disabled, then show a warning.
- If partial path checks don't pass and safety checks are strict, then fail.

This is implemented in https://github.com/trezor/trezor-firmware/commit/06de0743c81db723a84da4e2d0efe257080ecd54 (totally untested).

Nevertheless all of the listed solutions seem valid to me. Since I don't have good knowledge of the intricacies involved in the path checks, I would welcome input from @matejcik and @prusnak in choosing the best solution.

tsusanka, 2021-06-15:
I have a strong preference to go with Option 2 because:
1. Our partners won't be very happy if any warning persists. I would be benevolent in this and allow it for now.
2. It will be consistent with model T.

And of course it is an easy solution. Any objection from anyone?

matejcik, 2021-06-16:
I agree with option (2) as the simple way forward.

The reasoning behind full checks for `GetAddress` and partial checks for `SignTx` is as follows:
- the attack on `SignTx` is "provide mismatched `coin_name`, sign BTC transaction while showing LTC". The partial check stops this by checking the `coin_type` field, and avoids the majority of compatibility issues.
- the attack on `GetAddress` is "provide weird `address_n`, hold received funds for ransom". We want to warn the users if the path deviates from what we expect in any way.

TT's path checking is much stricter than this, but I see no particular reason to start enforcing this on T1 before the port. So I would keep the distinction as implemented.

IOW, let's explicitly add UC, Casa and GreenAddress support to `coin_known_path_check`.

(Let me also note that TT handling conflates several different "danger levels": arguably, `m/44'/0'/101'/0/0` is very different from `m/44/0/0/0/0` or `m/24684621/516582/5156/21/856/3548`, and _spending from_ `m/2468/13579/0/0/0/0/1000` is very differen than _showing address_ for the same path. This might be worth revisiting.)

mmilata, 2021-06-17:
Whoever implements the solution, please remove the `skip_t1` markers from test cases added in #1670.

andrewkozlik, 2021-06-21:
Thank you @tsusanka and @matejcik, you make good points so I will go ahead with Option 2. In particular I find @matejcik's reasoning for handing path checks differently in `SignTx` and `GetAddress` convincing and agree that we should revisit this topic in TT.

tsusanka, 2021-07-16:
Closed by https://github.com/trezor/trezor-firmware/pull/1711/commits/a6390e4b824f9405ecd5513155f0a094885c210a.

closed 2021-07-16 by tsusanka