trezor/trezor-firmware #1665: Add device tests for non-standard paths

andrewkozlik, 2021-06-10:
We have repeatedly had issues with breaking applications that use non-standard paths (Unchained Capital, Casa and GreenAddress), see below. A set of device tests is required in order to avoid this in the future. The tests should cover all messages that have an `address_n` field, but most importantly `GetAddress` and the transaction signing messages.
https://github.com/trezor/trezor-firmware/blob/74cf309a9311ee6ec33cea6f7b7bab1c5a15aedf/core/src/apps/bitcoin/keychain.py#L48-L62

closed 2021-06-17 by mmilata

bosomt, 2021-07-04:
@andrewkozlik is there anything that can/need to be tested on our side with new `2.4.1 `or `1.10.2` firmware ?

andrewkozlik, 2021-07-07:
> @andrewkozlik is there anything that can/need to be tested on our side with new `2.4.1 `or `1.10.2` firmware ?

This issue does not need QA.

PR #1670 (2021-06-17): test: add device tests for non-standard paths
Fixes #1665.