trezor/trezor-firmware #1667: cardano implementation in bip32.c

soooneee, 2021-06-11:
**Describe the bug**
There is no confirmation that kL is invalid if kL is divisible by the base order n in `hdnode_private_ckd_cardano()`

**Firmware version and revision**
X

**Desktop/smartphone setup (please complete the following information):**
X

**To Reproduce**
X

**Expected behavior**
X

**Screenshots**
X

**Additional context**
As far as I know `hdnode_private_ckd_cardano()` follows the [slip23](https://github.com/satoshilabs/slips/blob/master/slip-0023.md). Slip23's child key derivation algorithm is BIP32-Ed25519 ([ieee](https://ieeexplore.ieee.org/document/7966967), [google docs](https://drive.google.com/file/d/0ByMtMw2hul0EMFJuNnZORDR2NDA/view?resourcekey=0-5yJh8wV--HB7ve2NuqfQ6A)). 

The paper said, "If kL is divisible by the base order n, discard the child" in the section of C.Private child key of "V. BIP32-ED25519: SPECIFICATION". But I counldn't find this verification in the function.

refi93, 2021-06-11:
The code for that function in trezor-firmware:

https://github.com/trezor/trezor-firmware/blob/f93a8514e8e25b17b02360d9955c0a30999adf68/crypto/bip32.c#L292

is AFAIK taken straight from the cardano implementation:

https://github.com/input-output-hk/cardano-crypto/blob/04c211d7f9504b04eb6595cab62eb39a573467f1/cbits/encrypted_sign.c#L308

So I'd recommend checking there and approaching them directly as they should be the ones able to clarify it

tsusanka, 2021-06-21:
Corresponding issue: https://github.com/input-output-hk/cardano-crypto/issues/78

soooneee, 2021-06-22:
I got a reply from cardano : https://github.com/input-output-hk/cardano-crypto/issues/78
It will not be happen mathematically.

closed 2021-06-22 by soooneee

vincenthz, 2021-06-22:
some further rationale are noted at https://github.com/typed-io/rust-ed25519-bip32/blob/master/src/derivation/mod.rs#L80-L88