trezor/trezor-firmware #1674: Intermediary FW asks for FW confirmation on new devices

tsusanka, 2021-06-20:
If I install Intermediary 1.10.0 on brand new device from the factory (BL 1.4.0) I am prompted for firmware update confirmation which goes against the intermediary [spec](https://www.notion.so/satoshilabs/Intermediary-FW-3815baff0d4d4682b40dc7a3bcefbb9a). The spec says there should no confirmation needed on user's side.

This is what Trezor looks like after Intermediary FW is installed:
[image]

If I install normal FW afterwards this is what I get:

[image]

But the FW should be installed right away without any confirmation because the intermediary FW should delete itself. It should therefore equal to a device on which I run `trezorctl wipe-device --bootloader`, which looks like this and does not prompt before FW install:

[image]

I believe this is not a blocker for Suite's release - there is simply just an extra confirmation which is not ideal but not blocking as Suite handles it. It would be nice to fix this though so the extra confirmation is removed.

Note that as described in the spec the Intermediary FW **can not erase storage for initialized devices**. Having prompt there is OK as written in the spec.

Full video [here](https://drive.google.com/file/d/10JxEPD1nv5cGpYH1z5YgF-O7kIv_kwAG/view?usp=sharing).

prusnak, 2021-06-20:
The issue here is that bootloader 1.10.0 still detects the firmware via `firmware_present_new`

https://github.com/trezor/trezor-firmware/blob/69d1465e0862aebf52640bf6f2c95a3a486df0ef/legacy/bootloader/signatures.c#L124-L137

although we invalidate it here:

https://github.com/trezor/trezor-firmware/blob/69d1465e0862aebf52640bf6f2c95a3a486df0ef/legacy/intermediate_fw/trezor.c#L73-L85

hiviah, 2021-06-21:
I'd guess the word is not written either because the unlock is not correct or needs erase first. There was something with using `memory_write_unlock` (hack that might be needed just for replacing bootloader) or missing erasing the sector first.

prusnak, 2021-06-21:
You don't need to erase the sector if you just flip `1`s to `0`s, which is what we do here. But you might be right about write_unlock.

hiviah, 2021-06-21:
Possibly we will need to revert back to erasing the first firmware sector.

hiviah, 2021-06-21:
Seems that we can't just erase first sector, we need to write at least part of header with signatures back:

* [FW signatures on flash are required to avoid storage erase](https://github.com/trezor/trezor-firmware/blob/69d1465e0862aebf52640bf6f2c95a3a486df0ef/legacy/bootloader/usb.c#L386-L407)
* [Header is at 1st kB of first 64 kB FW block](https://github.com/trezor/trezor-core/blob/master/docs/bootloader.md#firmware-header)
* first FW sector is `Sector  4 | 0x08010000 - 0x0801FFFF` (see `memory.h`)

STM32F205 flash programming manual though explicitly states that when flashing, flipping 1 to 0 should be possible without erase first. At the same time there can be unspecified occurence when it won't be written.

prusnak, 2021-06-21:
> At the same time there can be unspecified occurence when it won't be written.

Maybe we can investigate this more?

Maybe add nop-nop-nop or HAL_Delay after `_flash_program_word(FLASH_FWHEADER_START, 0);` 

Maybe delete more bytes than first 4, but stil not corrupt the signatures?

hiviah, 2021-06-21:
I tested the erasing of first 4 bytes and it works.

This seems to be the issue - if storage is initialized, the `invalidate_firmware` is not executed, only reboot with 'stay in firmware' flag:

https://github.com/trezor/trezor-firmware/blob/69d1465e0862aebf52640bf6f2c95a3a486df0ef/legacy/intermediate_fw/trezor.c#L145

hiviah, 2021-06-21:
Fix in PR #1675.

hiviah, 2021-06-22:
It seems we can't get rid of the confirmation without changing bootloader, detailed explanation in this PR comment:

https://github.com/trezor/trezor-firmware/pull/1675#issuecomment-865940901

Quoted:

> I tried to prove that changing the 4 magic bytes won't change the outcome, but actually proved the opposite.
> 
> Because of this FW magic check
> 
> https://github.com/trezor/trezor-firmware/blob/f39abc8356d18a50e3201f534cb22c3ca72b5c0f/legacy/bootloader/usb.c#L233
> 
> the variable `old_was_signed` remains false (`SIG_FAIL`) and this following check on it will cause firmware erase:
> 
> https://github.com/trezor/trezor-firmware/blob/f39abc8356d18a50e3201f534cb22c3ca72b5c0f/legacy/bootloader/usb.c#L391
> 
> So we can't get rid of the confirmation without changing bootloader.

tsusanka, 2021-06-30:
So that's the final verdict right? So I guess we'll close as a `won't fix`?

hiviah, 2021-06-30:
I though long about a workaround, and possibly having `if(firmware_present_new() || firmware_was_invalidated())` in the first check, where `firmware_was_invalidated()` would check for 32-bit zero instead of FW magic, should probably fix it, along with the PR.

However I can't really be 100% sure since it's really not testable until the new bootloader is put into intermediate FW and whole thing signed, also not 100% sure it won't create some hole (the bootloader code is hard to understand with all legacy hacks). So maybe some other pair of eyes should review if this makes sense, otherwise we'd have to rollback the entire thing if we find out it'd be blocked on another check.

tsusanka, 2021-07-01:
@prusnak what do you think? We definitely do not want to make the bootloader even more complicated. So if this sounds hacky, we'll solve it on Suite side.

prusnak, 2021-07-01:
I like the suggestion from hiviah and I would implement and merge it. But I would not block the development with Suite and use the current 1.10.0 bootloader. We can later release this change in further bootloader update.

tsusanka, 2021-07-01:
Definitely! Ok @hiviah will you add this workaround to your PR pls?

prusnak, 2021-07-01:
@hiviah maybe we can just do the following:

* expand function `firmware_present_new(void)` to `firmware_present_new(bool allow_invalidated)`
* and then use `firmware_present_new(false)` or `firmware_present_new(true)` correspondingly

Will you please prepare a PR for this?

tsusanka, 2021-07-19:
I made an issue in Suite: #4047. So should we mark this as `won't fix` and close? :thinking:

hiviah, 2021-07-19:
I'd say yes since the fix is really invasive given the issue - see PR https://github.com/trezor/trezor-firmware/pull/1693

tsusanka, 2021-07-19:
Fine, I agree. Let's reopen if needed but I really think it is not worth it.

closed 2021-07-19 by tsusanka