trezor/trezor-firmware #1686: Investigate invalid Monero address after Trezor firmware update with Shamir backup

ph4r05, 2021-07-01:
**Describe the bug**
Some user reports that when recovering Trezor seed (with Shamir backup) and/or updating Trezor firmware that computed Monero address is different from address they used before restore and/or update.

More info below.

**Firmware version and revision**
Observed on firmware versions 2.3.6, 2.4.0.
Monero wallet 0.17.2.0 (the newest)

**Desktop/smartphone setup:**
 - Wallet software: Trezor Suite, Monero-GUI wallet
 - OS: multiple OS, reports from OSX

**To Reproduce**
1. create monero wallet
2. recover Shamir backup seed, update Trezor Firmware
3. open monero wallet -> invalid address (computed and stored do not match)

**Expected behavior**
Monero addresses should be the same before and after seed recovery and firmware update.

**Additional context**
As discussed with @tsusanka over IM, this issue is created to track progress on reported problems from reddit:
- https://www.reddit.com/r/monerosupport/comments/obb6uu/issues_accessing_monero_wallet_using_gui_after/
- https://old.reddit.com/r/monerosupport/comments/n0s3w0/using_shamir_on_trezor_and_passphrase_stopped/
- https://old.reddit.com/r/monerosupport/comments/no8na1/trezor_model_t_help/
- https://www.reddit.com/r/monerosupport/comments/neofbx/trezor_firmware_236_device_wallet_address_does/

Also other people in comments mention they have a similar problem. The most probable reason can be invalid passphrase (casing, spaces), as different passphrase leads to different (and valid) Monero addresses. Also, some people were confused that they can use Bitcoin without passphrase and Monero with a passphrase on the same Trezor device. Thus one could just forgot the exact passphrase.

Anyway, it would be good to investigate this problem a bit further as it could be quite critical bug. 
Some users mention using Shamir backup. We can focus on several scenarios:
- firmware update, with and without passphrase
- Shamir backup restore, with and without passphrase
- Shamir backup and firmware update, with and without passphrase

If we cannot reproduce the problem during manual testing, the problem can be only with a small portion of seed space. (For example. if first or last bytes are 0x00). 

So I would suggest then doing some automated testing, like:
- Generate 2^18 = 262 144 random seeds (16 for 2 bytes being null randomly, 2 additional bits for more samples)
- Store corresponding Monero addresses
- Update firmware (or run update procedure for seed recovery), recover seed with and without shamir
- Check if current Monero address corresponds to the old one for the same seed 

### Checks performed
I did some code review of credentials-related code, both codebase and recent changes to both Trezor Firmware and Monero and found nothing obviously suspicious.

I also did the following test (no Shamir):  
- Initialized Trezor with a testing seed and loaded Monero address loaded by a dev python script. The same test was performed on firmware versions 2.4.0 (latest dev version) and 2.1.8 (older version, Oct 2019). Address was `497MP7bh5MkCZ1TiZZoYi84s39n5V5HpSgPvrqqUMqw6XcWDu4QhBs4VkqiPsNMhTUPAHHE6DUo9UUqpfSZfxkANPw44Fgn`. Thus generating address from the same seed did not change between those versions.
- I used the initialized trezor with `monero-wallet-cli`, versions 0.17.2.0 (current) and 0.16.0.1 (July 2020). Both versions returned the same address `497MP7bh5MkCZ1TiZZoYi84s39n5V5HpSgPvrqqUMqw6XcWDu4QhBs4VkqiPsNMhTUPAHHE6DUo9UUqpfSZfxkANPw44Fgn`. Thus address generation is the same between these versions.
- I repeated tests with `monero-wallet-cli` with passphrase `a`, both wallet versions returned `4BBKEeg8iH3JtyQKfdh5KRYNe2WK4aDMBeMwPvkjrm45BQ8bVHurmLyadDx3EiM6AjNH7JJx5TMNrjC4JLZEhszc5f3G8Yg`.
- Test was repeated also with old Trezor firmware 2.1.8, passphrase `a` returned `4BBKEeg8iH3JtyQKfdh5KRYNe2WK4aDMBeMwPvkjrm45BQ8bVHurmLyadDx3EiM6AjNH7JJx5TMNrjC4JLZEhszc5f3G8Yg` address.

I thus cannot reproduce derivation problem by this naive test. 

The addresses are consistent with output from https://github.com/ph4r05/monero-agent/blob/master/PoC.md#seed-utility
```
python3 -m monero_poc.seed 'permit universe parent weapon amused modify essay borrow tobacco budget walnut lunch consider gallery ride amazing frog forget treat market chapter velvet useless topple' --slip0010

0010
Seed bip39 words: permit universe parent weapon amused modify essay borrow tobacco budget walnut lunch consider gallery ride amazing frog forget treat market chapter velvet useless topple
Seed bip32 b58:   019a688cb29a48235e6b92dc557798eab567e60916e60711b87fceb15f2e26b7d52066515c1d14fdba5b92e44ee33b74b0cfebba1d545e61b34822410dd76cdd

Seed Monero:      14821d0bc5659b24cafbc889dc4fc60785ee08b65d71c525f81eeaba4f3a570f
Seed Monero wrds: geek together unwind nostril tugs afraid both keyboard zoom oncoming vixen withdrawn napkin railway imagine friendly zebra candy deftly ferry aerial axle jaded light jaded

Private spend key: 14821d0bc5659b24cafbc889dc4fc60785ee08b65d71c525f81eeaba4f3a570f
Private view key:  a6ccd4ac344a295d1387f8d18c81bdd394f1845de84188e204514ef9370fd403

Public spend key:  c59186a85caae34510668d4eb7a71d1717f7d0dbea991feb877be781e9f851b7
Public view key:   049b7cf656b84fabeb88bce18f804f847c6e0dd57bf4bba67479458b162063cb

Mainnet Address:   497MP7bh5MkCZ1TiZZoYi84s39n5V5HpSgPvrqqUMqw6XcWDu4QhBs4VkqiPsNMhTUPAHHE6DUo9UUqpfSZfxkANPw44Fgn
Testnet Address:   9zetsNFxMirCZ1TiZZoYi84s39n5V5HpSgPvrqqUMqw6XcWDu4QhBs4VkqiPsNMhTUPAHHE6DUo9UUqpfSZfxkANPwbdsur
Stagenet Address:  59KPTxWeixrCZ1TiZZoYi84s39n5V5HpSgPvrqqUMqw6XcWDu4QhBs4VkqiPsNMhTUPAHHE6DUo9UUqpfSZfxkANPv8LFHX

python3 -m monero_poc.seed 'permit universe parent weapon amused modify essay borrow tobacco budget walnut lunch consider gallery ride amazing frog forget treat market chapter velvet useless topple' --slip0010 --passphrase 'a'

Seed bip39 words: permit universe parent weapon amused modify essay borrow tobacco budget walnut lunch consider gallery ride amazing frog forget treat market chapter velvet useless topple
Seed bip32 b58:   ca23c96aa8552adf211ebbe1d23f78670b5ffb541b5276e26db1b2c7943ae7b354b013805b1836390187a1a3d5915f109a0f7a67c4e4603bb451a5df12bb5931

Seed Monero:      0b86cf0e71204ca3cdc389daf0bb1cf654ac7d54edfed68a9faf921ba140a708
Seed Monero wrds: maul loudly nearby buffet hacksaw zones kernels edgy baffles match extra eclipse uphill arena hounded wobbly actress muppet pebbles onward rift tether scrub snake rift

Private spend key: 0b86cf0e71204ca3cdc389daf0bb1cf654ac7d54edfed68a9faf921ba140a708
Private view key:  3b1213f644062fbfb15c4fa35e656659e4105ac728b791ce5ad10af98bc6d200

Public spend key:  fc25f0a1a1fb4e6afe5ffa15cd06fcbb913eb55605d09adf5de735163d4f4a3e
Public view key:   2ba093948b429ec90729ecd0f705b87f36154612756433fc3dc7d8dbbf646529

Mainnet Address:   4BBKEeg8iH3JtyQKfdh5KRYNe2WK4aDMBeMwPvkjrm45BQ8bVHurmLyadDx3EiM6AjNH7JJx5TMNrjC4JLZEhszc5f3G8Yg
Testnet Address:   A2iriuLPze9JtyQKfdh5KRYNe2WK4aDMBeMwPvkjrm45BQ8bVHurmLyadDx3EiM6AjNH7JJx5TMNrjC4JLZEhszc5dqa8Po
Stagenet Address:  5BPMKVb6Mt9JtyQKfdh5KRYNe2WK4aDMBeMwPvkjrm45BQ8bVHurmLyadDx3EiM6AjNH7JJx5TMNrjC4JLZEhszc5eeaUED
```

Python utility to extract Monero address from Trezor is also in package https://github.com/ph4r05/monero-agent/blob/master/PoC.md#seed-utility
Can be used to get Monero address without using monero-wallet-cli. 

```
pip3 install monero_agent[poc,trezor]
python3 -m monero_poc.trezor_address  --trezor-path 'bridge:1'  # path from `trezorctl list` 

Please confirm action on your Trezor device.
<MoneroAddress: {'address': b'497MP7bh5MkCZ1TiZZoYi84s39n5V5HpSgPvrqqUMqw6XcWDu4QhBs4VkqiPsNMhTUPAHHE6DUo9UUqpfSZfxkANPw44Fgn'}>
```

ph4r05, 2021-07-01:
Update on the last report from reddit.

The issue ocurred once customer wiped his Trezor and recovered from Shamir. Same firmware (2.3.6).

Passphrase is allegedly disabled, as suite does not ask for passphrase. But this could be from wiping and setting no passphrase option. Trezor uses PIN.

Trezor Suite shows still the same BTC address, just monero changed.
One hypothesis is that wallet oroginally had a passphrase.

Monero wallet was 0.17.1.9 when creating the wallet, now 0.17.2.0.

ph4r05, 2021-07-02:
Another report with exact same problem https://www.reddit.com/r/monerosupport/comments/obvlt7/monero_gui_stating_that_trezor_t_address_is_not/

tsusanka, 2021-07-02:
Note that Suite turns on passphrase by default even when the passphrase feature is not being used. So if one recovers Trezor in Suite passphrase will be turned on. So the device most likely has passphrase enabled. Maybe we can try if `Shamir & no passphrase` == `Shamir & empty passphrase`?

tsusanka, 2021-07-02:
I tested this and the addresses were correct.. Full log:

<details>
tsusanka@tsusanka ~/> trezorctl get-features                                    
Features (168 bytes) {
    backup_type: Bip39 (0),
    capabilities: [
        Bitcoin (1),
        Bitcoin_like (2),
        Binance (3),
        Cardano (4),
        Crypto (5),
        EOS (6),
        Ethereum (7),
        Lisk (8),
        Monero (9),
        NEM (10),
        Ripple (11),
        Stellar (12),
        Tezos (13),
        U2F (14),
        Shamir (15),
        ShamirGroups (16),
        PassphraseEntry (17),
    ],
    device_id: 'CB916436AE3C744E355C8175',
    flags: 0,
    initialized: True,
    label: '',
    language: 'en-US',
    major_version: 2,
    minor_version: 4,
    model: 'T',
    needs_backup: False,
    no_backup: False,
    passphrase_always_on_device: False,
    passphrase_protection: False,
    patch_version: 1,
    pin_protection: False,
    recovery_mode: False,
    revision: 20 bytes 0x47c2be9f1386a2a0c9247fcdf04c1358d848bf00,
    sd_card_present: True,
    sd_protection: False,
    unfinished_backup: False,
    unlocked: True,
    vendor: 'trezor.io',
    wipe_code_protection: False,
}
tsusanka@tsusanka ~/> trezorctl monero get-address -n "44'/128'/0'"             
44tjw4eUqTs1ZJJsXJr94FFdsyEuQ1AarCncqcDzy5epS8HRdLYmRQMhabCKAVxHfN8W1AhDpN5NwLN338FcAu1J5hu13HE
tsusanka@tsusanka ~/> trezorctl set passphrase enabled                          
Please confirm action on your Trezor device.
Settings applied
tsusanka@tsusanka ~/> trezorctl monero get-address -n "44'/128'/0'"             
Please confirm action on your Trezor device.
44tjw4eUqTs1ZJJsXJr94FFdsyEuQ1AarCncqcDzy5epS8HRdLYmRQMhabCKAVxHfN8W1AhDpN5NwLN338FcAu1J5hu13HE
tsusanka@tsusanka ~/> trezorctl get-features                                    
Features (168 bytes) {
    backup_type: Bip39 (0),
    capabilities: [
        Bitcoin (1),
        Bitcoin_like (2),
        Binance (3),
        Cardano (4),
        Crypto (5),
        EOS (6),
        Ethereum (7),
        Lisk (8),
        Monero (9),
        NEM (10),
        Ripple (11),
        Stellar (12),
        Tezos (13),
        U2F (14),
        Shamir (15),
        ShamirGroups (16),
        PassphraseEntry (17),
    ],
    device_id: 'CB916436AE3C744E355C8175',
    flags: 0,
    initialized: True,
    label: '',
    language: 'en-US',
    major_version: 2,
    minor_version: 4,
    model: 'T',
    needs_backup: False,
    no_backup: False,
    passphrase_always_on_device: False,
    passphrase_protection: True,
    patch_version: 1,
    pin_protection: False,
    recovery_mode: False,
    revision: 20 bytes 0x47c2be9f1386a2a0c9247fcdf04c1358d848bf00,
    sd_card_present: True,
    sd_protection: False,
    unfinished_backup: False,
    unlocked: True,
    vendor: 'trezor.io',
    wipe_code_protection: False,
}
</details>

prusnak, 2021-07-02:
> I tested this on master

Did you try disconnecting the device before enabling the passphrase? That's the only way we can be sure we are not interfering with some caching mechanism.

tsusanka, 2021-07-02:
Same results on 2.3.6 and 2.4.0. I have tested on the emulator. I'll test now on actual device with disconnects in between :+1:.

tsusanka, 2021-07-02:
Same results on 2.3.6 and 2.4.0 on device. I even checked the address in Monero GUI v0.17.2.0 and always received the same address.

ph4r05, 2021-07-02:
@tsusanka thanks for info! Just for completeness, was the firmware was updated by the suite during the test? It seems this is common feature of the bug reports. Thanks!

tsusanka, 2021-07-02:
@ph4r05 it was not, I did it via `trezorctl`. I guess we can retest that way, but the only difference should be that Suite will turn on the passphrase (which I did manually via trezorctl).

I am really wondering if this is not caused by the simple fact that users did not know anything about passphrase, we suddenly turned it on for them (via Suite) and they might have confused passphrase with the Monero wallet password.

Anyway @andrewkozlik will also have a look and I will instruct QA to test using Suite as well.

tsusanka, 2021-07-02:
@sorooris please someone from QA test this following scenario:
1. Install 2.3.6 via `trezorctl`
2. Initialize Shamir Backup
3. Ask for address with empty passphrase in Monero GUI
4. Wipe device
5. Install 2.4.1 via Suite
6. Recover using the Shamir Backup
7. Ask for address with empty passphrase in Monero GUI

sorooris, 2021-07-02:
> 5. Install 2.4.1 via Suite

@tsusanka did you mean **2.4.0** via Suite? **2.4.1** isn't offered by Suite yet.

Anyway, I followed the steps above and both the primary and the secondary receive addresses **were the same after recovering** first from the local file and second from the device so I couldn't replicate the issue:
2.3.6 b19cbf67 - new wallet | 2.4.0 ea3596ad wallet - recovered from local file | 2.4.0 ea3596ad - wallet recovered from device
 -|-|- 
[image] | [image] | [image]

I can provide the full addresses and the Shamir recovery seeds upon request. 

@wendys-cats is in the process of testing the same steps as above but with 2.4.1 firmware.

Please let me know if we should test any other cases.

ph4r05, 2021-07-02:
@selsta pls, if you notice further bug reports related to this issue on reddit, pls link it here. Thanks!

wendys-cats, 2021-07-02:
**2.3.6 to 2.4.1 test**

- address tested against trezorctl and Monero GUI
- Shamir backup with empty passphrase

**Info:**
Device TT with 2.3.6 regular public FW and 2.4.1 revision 0x01c1ae426fe911619f0312fa417ecdd7e102848c FW
Monero GUI Walet 0.17.2.0 GUI version 0.17.2.2-937cb98 (Qt 5.15.2)
OS Windows 10 Version 10.0.19042.1052
Trezorctl 0.12.2

**Proccess:**
1. Install 2.3.6 regular FW into TT device via trezorctl
2. Recover Shamir backup via Suite
3. Get Monero address from trezorctl 
4. Get Monero address from Monero GUI via Create new wallet with HW device (empty passphrase)
5. Wipe device
6. Install 2.4.1 regular revision 0x01c1ae426fe911619f0312fa417ecdd7e102848c via trezorctl
7. Recover Shamir backup via Suite
8. Get Monero address from trezorctl 
9. Get Monero address from Monero GUI via file
10. Get Monero address from Monero GUI via Create new wallet with HW device (empty passphrase)
11. Compare addresses

**Result:** 
All the addresses were identical. I was not able to reproduce the issues.
Can provide addresses and shamir backup seed upon request.

andrewkozlik, 2021-07-07:
I haven't been able to reproduce this issue on Linux and I don't see how this could be related to Shamir backup, but I have a hypothesis that's worth investigating. I think there might be some bug in how versions `0.17.2.*` of `monero-wallet-gui` handle the passphrase. This bug might be manifesting differently on Linux, MacOS and Windows. On Linux `monero-wallet-gui` versions 0.17.2.0, 0.17.2.1 and 0.17.2.2 crash randomly when I try to enter a non-empty passphrase on the computer, usually with `double free or corruption (out)` or `corrupted size vs. prev_size`. Other than that they seem to work fine. I would like to see what happens on MacOS.

@qa please install Trezor firmware 2.3.4, generate a seed and try this on a MacOS:

For the following versions of `monero-wallet-gui`:
- v0.17.2.2
- v0.17.1.9

and for the following cases:
- passphrase disabled on Trezor (via Wallet or Suite)
- empty passphrase entered on Trezor
- empty passphrase entered on the computer
- passphrase `a` entered on Trezor
- passphrase `a` entered on the computer

take the following steps:
1. `rm -r ~/Monero/`
2. Run `monero-wallet-gui`.
3. Create new wallet with HW device.
4. Set passphrase as above.
5. Note the primary address of Account #0.

Check that:
- an empty or disabled passphrase always produce the same primary address on both versions of `monero-wallet-gui`,
- passphrase `a` produces the same primary address regardless of the `monero-wallet-gui` version or the mode of entry.

ph4r05, 2021-07-09:
@andrewkozlik interesting, I replicated the crash on computer-entered passphrase on v0.17.2.2 on Linux. I've created an issue for this  https://github.com/monero-project/monero/issues/7778.

I am now trying to replicate this on the master branch (building).

The crash trace looks innocent with respect to the passphrase entry, destructor of the PassphraseAck message tries to free already freed passphrase buffer. This should not affect already sent passphrase. Also, the app should crash and not allow to continue creating a wallet. Trezor now also shows passphrase entered on the host, right?

andrewkozlik, 2021-07-09:
> Trezor now also shows passphrase entered on the host, right?

Correct.

To be honest I can't make sense of the issue, so I am just guessing at what could be the problem. There were two posts that caught my attention:
- https://www.reddit.com/r/monerosupport/comments/n0s3w0/using_shamir_on_trezor_and_passphrase_stopped/?utm_source=reddit&utm_medium=usertext&utm_name=monerosupport&utm_content=t1_h03y41m
userthrowaway143hu comments that "It had worked the first few times it was set up and used to receive some coins. After that, the passphrase only worked if typed on the device, not the computer." That seems to indicate some issue with the monero-gui. Of course the user might have just been making a typo while typing on the computer keyboard. It doesn't explain why the passphrase recently "stopped working" altogether.
- https://old.reddit.com/r/monerosupport/comments/obb6uu/issues_accessing_monero_wallet_using_gui_after/
This user claims they didn't have a passphrase. They recovered their wallet through Suite, which means that passphrase was automatically enabled for them after recovery, even if they weren't using one before. Rolling back the firmware and monero-gui versions doesn't solve the problem, but there still seems to be one difference: before the passphrase was apparently not enabled and now it is. Maybe they should try disabling it?

ph4r05, 2021-07-09:
@andrewkozlik agree, I am on the same page, just generating various hypotheses. I don't see any obvious reason why there should be a bug after reviewing the code and trying to replicate it (no one replicated it so far). 

These 2 are interesting, trying to disable passphrase is also an option, however, now I am trying to explain them to use `trezorctl` to generate Monero address, so monero-wallet is eliminated from the "equation" and they can quickly try different passphrases. I also like the idea they may have confused passphrase and monero wallet password.

What do you think abot hypothesis it can be non-deterministic bug? Originally I proposed to generate 2^20 seeds and try to automate shamir recovery and address generation with passphrase. Do you think it makes sense?

andrewkozlik, 2021-07-09:
> What do you think abot hypothesis it can be non-deterministic bug? Originally I proposed to generate 2^20 seeds and try to automate shamir recovery and address generation with passphrase. Do you think it makes sense?

It seems unlikely to me, because the users are not experiencing anything similar with Bitcoin and the node derivation mechanism in Monero is the same, isn't it? It also seems that these issues started popping up just in the past 2 months or so, which would indicate that it's related to some sw/fw update. But who knows, I guess anything is worth a shot at this point.

ph4r05, 2021-07-09:
Another hypothesis we are testing with users - they may used old seed with the Monero wallet, then migrated to Shamir seed and forgot about the Monero wallet using old seed. Will know more later.

I inspected the crash, it seems it was introduced  20 Jan, so older versions should not be affected by this. As said before, it crashes, so wallet is not usable after entry. And also Trezor shows the passphrase for verification, so I see this bug as a culprit quite unlikely.

dEBRUYNE-1, 2021-07-11:
Adding for visibility purposes. A user that seemingly incurred a similar issue, but managed to resolve it themselves.

https://www.reddit.com/r/monerosupport/comments/ohfegp/help_cant_access_my_gui_wallet_trezor_model_t/h4q4z2b/

ph4r05, 2021-07-11:
@andrewkozlik @tsusanka this is very interesting, the guy resolved his issue: https://www.reddit.com/r/monerosupport/comments/obb6uu/issues_accessing_monero_wallet_using_gui_after/h4ua6cl/?utm_source=share&utm_medium=ios_app&utm_name=iossmf&context=3

What do you suggest? Thanks!

andrewkozlik, 2021-07-11:
"for some reason this changed my adress wallet" - this is confirming my suspicion that the monero-gui is handling the passphrase incorrectly, i.e. inconsistently when the passphrase is disabled or enabled and entered on Trezor or on the computer. On Linux I haven't been able to demonstrate this. Can somebody try this on a MacOS, which is what user 93497643455437 is using. In the first place we need to be able to reproduce this.

"Upgrading to Trezor firmware 2.4.0 and rerunning the wipe/restore process (as described above) resulted in successful recovery." - this indicates to me that the issue does not have to do with the firmware.

ph4r05, 2021-07-11:
I am not able to reproduce this on OSX, build 9c18f2767bc62796eacb0428ac50abf8fc5c14dd:
- Tested with empty passphrase entered both on PC and Trezor
- Tested with Trezor initialized without passphrase with the same seed, monero generated the same address

Tested on monero-wallet-cli, GUI uses cli backend, but we can check it as well.

ph4r05, 2021-07-11:
I've just tested monero-gui on the master (1c61ec2511af8a4c545c1ff3791668becdf33dbe) with same scenario. always got the address `497MP7bh5MkCZ1TiZZoYi84s39n5V5HpSgPvrqqUMqw6XcWDu4QhBs4VkqiPsNMhTUPAHHE6DUo9UUqpfSZfxkANPw44Fgn`

andrewkozlik, 2021-07-12:
> I've just tested monero-gui on the master (1c61ec2511af8a4c545c1ff3791668becdf33dbe) with same scenario. Always got the address `497MP7bh5MkCZ1TiZZoYi84s39n5V5HpSgPvrqqUMqw6XcWDu4QhBs4VkqiPsNMhTUPAHHE6DUo9UUqpfSZfxkANPw44Fgn`

I think it's always better to try with the GUI, because the issue could be coming from the QT UI components, e.g. something like the dialog might be returning the string with a newline, which will result in different strings on Linux/Mac/Windows. Also I think it's better to try reproducing this with a released version, rather than building from master, because it can be related to some library version that the code was compiled with. Try testing with v0.17.2.0, which is what user 93497643455437 claims to be using. While you are at it, I'd also try v0.17.1.9.

ph4r05, 2021-07-12:
@andrewkozlik I've just tested both v0.17.2.0, v0.17.1.9 released on github on OSX, same scenario, both trezor passphrase enabled/disabled in the device (tested on real device as release version does not support emulator), always getting the same result with empty passphrases.

andrewkozlik, 2021-07-12:
@ph4r05 were you using Shamir backup? I think you don't necessarily have to do Shamir recovery, just generate a seed for Shamir backup.

ph4r05, 2021-07-12:
@andrewkozlik no, I am using normal seed initialization, without Shamir. Is there a way to initialize Trezor built with debug mode on from CLI?

andrewkozlik, 2021-07-12:
> @andrewkozlik no, I am using normal seed initialization, without Shamir. Is there a way to initialize Trezor built with debug mode on from CLI?

For debug mode you need a special build of Trezor firmware. Maybe you meant initialize with Shamir backup? That would be `trezorctl device setup -b shamir`

ph4r05, 2021-07-12:
I have a debug build so I can initialize the Trezor always with the same seed and passphrase enabled/disabled from PC. Is it possible to setup shamir backup wallet completely from CLI with fixed seed? I am quite busy (and lazy) to enter shamir seeds via Trezor to test this.

matejcik, 2021-07-12:
you can use `trezorctl device load` and specify multiple `-m <your mnemonic here>` options for individual shares

andrewkozlik, 2021-07-12:
> Is it possible to setup shamir backup wallet completely from CLI with fixed seed?

I don't think there is a way to do that from trezorctl. Is there @matejcik? But you can use this python function: https://github.com/trezor/trezor-firmware/blob/f8bb90366addcce3218d65e723b4185aee3b0df8/tests/common.py#L116

> I have no pasphrase enabled now and have no idea how to enable it without wiping the Trezor and recovering from shares.

`trezorctl set passphrase enabled`
You can also do it in Suite or Wallet by going to advanced settings.

ph4r05, 2021-07-12:
Thanks, this worked: `trezorctl set passphrase enabled`! However, I could not replicate the problem.

ph4r05, 2021-07-12:
Another idea - as I am using debug build, I do no restart Trezor between tests as restart wipes the secret. Could be cache of some sort (credentials, passphrase) a culprit?

andrewkozlik, 2021-07-12:
> Could be cache of some sort (credentials, passphrase) a culprit?

I don't believe so.

This is super weird. I am beginning to contemplate whether it's some bizarre user error. Namely:
> 10. Monero GUI now prompts for me to enter passPHRASE into computer or hardware. I enter the passphrase.

Makes me want to see a video of this.

matejcik, 2021-07-12:
Just to bring some order to the issue, here's a full list of Reddit users who claimed problems.

> https://www.reddit.com/r/monerosupport/comments/obb6uu/issues_accessing_monero_wallet_using_gui_after/

user **93497643455437**
was using passphrase, it was disabled after restore, re-enabling and entering passphrase solved the problem (see comment above)

---

> https://old.reddit.com/r/monerosupport/comments/n0s3w0/using_shamir_on_trezor_and_passphrase_stopped/

user **userthrowaway143hu**

writes:
> in fact I normally I don't use a passphrase at all but thought I'd start seeing how the Model T felt with that extra layer.
> (...)
> so I used the same passphrase as the wallet password

Seems that they didn't use passphrase, tried to enable, entered a non-empty one.
waiting for confirmation [here](https://www.reddit.com/r/monerosupport/comments/n0s3w0/using_shamir_on_trezor_and_passphrase_stopped/h4x16br/?utm_source=reddit&utm_medium=web2x&context=3)

---

> https://old.reddit.com/r/monerosupport/comments/no8na1/trezor_model_t_help/

user **subsilent**
no indication they use Shamir
per this thread, likely simply a wrong passphrase?

---

> https://www.reddit.com/r/monerosupport/comments/neofbx/trezor_firmware_236_device_wallet_address_does/

user **atriker**
not many details, no indication Shamir is used

in the same thread, also user **DropOfTheDay** claims the same problem:
https://www.reddit.com/r/monerosupport/comments/neofbx/trezor_firmware_236_device_wallet_address_does/gysa8cd/?utm_source=reddit&utm_medium=web2x&context=3
also no details, no indication Shamir is used

---

> https://www.reddit.com/r/monerosupport/comments/obvlt7/monero_gui_stating_that_trezor_t_address_is_not/

user **ChestBusy**
wrote down "passphrase" but does not indicate whether that's the wallet password or Trezor passphrase
waiting for details [here](https://www.reddit.com/r/monerosupport/comments/obvlt7/monero_gui_stating_that_trezor_t_address_is_not/h4vz1l0/?utm_source=reddit&utm_medium=web2x&context=3)

---

> https://www.reddit.com/r/monerosupport/comments/ohfegp/help_cant_access_my_gui_wallet_trezor_model_t/h4q4z2b/

user **Thornet77**
disabled passphrase, problem went away. I am waiting for clarification [here](https://www.reddit.com/r/monerosupport/comments/ohfegp/help_cant_access_my_gui_wallet_trezor_model_t/h4wzf5d/?utm_source=reddit&utm_medium=web2x&context=3)

---

In summary, so far it seems it's a veritable collection of user errors.

93497643455437, 2021-07-12:
> This is super weird. I am beginning to contemplate whether it's some bizarre user error. Namely:
> 
> > 10. Monero GUI now prompts for me to enter passPHRASE into computer or hardware. I enter the passphrase.
> 
> Makes me want to see a video of this.

Hey Andrew,

I'm the user 93497643455437 who was able to resolve their issue after restoring from wallet.trezor.io and then enabling an empty passphrase. I'd be happy to record a video (if possible without doxing myself) of what I encountered if you provide more details on what exactly you'd like to see.

rating89us, 2021-07-13:
It seems to me that part of problem is that users are wiping Trezor and disabling passphrase on the device, and after that they're opening their previous Monero GUI wallet file expecting that this wallet file will detect whether passphrase functionality is enabled or disabled on the recently wiped Trezor device.

Maybe when creating Monero GUI wallet file for the first time, Monero GUI should store whether this file corresponds to a Trezor with or without passphrase enabled. This way, in the future, if the user tries to open a "passphrase enabled wallet file" using a "passphrase disabled Trezor", Monero GUI should warn the user: "You're opening a wallet that was created with a passphrase enabled device, but now you are using a device with passphrase disabled. Please open Trezor Suite and enable passphrase for your device in order to open this wallet."

rating89us, 2021-07-14:
About https://www.reddit.com/r/monerosupport/comments/ohfegp/help_cant_access_my_gui_wallet_trezor_model_t/h4q4z2b/

It seems that the opposed happened.

The user had a "passphrase-disabled wallet file" and always used a "passphrase-disabled Trezor".

He then reported that he had to enable passphrase on Trezor suite. Probably he opened Monero GUI and tried to use his "passphrase-disabled wallet file" with his now "passphrase-enabled Trezor", and got the error "Device wallet does not match wallet adress". The solution was to open Trezor Suite and disable passphrase on his Trezor device.

ph4r05, 2021-07-14:
@rating89us the thing is that an empty passphrase and disabled passphrase generates the same wallet addresses. 

We were not able to replicate the problem. I already tried to generate wallet with pssphrase disabled, then opened it with empty passphrase successfully. Also reversed (empty passphrase, opened with disabled pssphrase Trezor).

andrewkozlik, 2021-07-14:
> Hey Andrew,
>
> I'm the user 93497643455437 who was able to resolve their issue after restoring from wallet.trezor.io and then enabling an empty passphrase. I'd be happy to record a video (if possible [...]

Hi @93497643455437, thanks for helping to diagnose this issue. I was under the impression that you did not have a passphrase set up, but after @matejcik's comment and after re-reading your posts, I am in doubt. When you say:
> 10. Monero GUI now prompts for me to enter passPHRASE into computer or hardware. I enter the passphrase.

what does "enter the passphrase" mean? Are you entering an empty passphrase, i.e. just clicking OK, without any input, or are you actually typing in some non-empty passphrase? While we are at it, does it work the same if you choose entry on the computer vs. if you enter it on the Trezor?

rating89us, 2021-07-14:
> @rating89us the thing is that an empty passphrase and disabled passphrase generate the same wallet addresses.
> 
> We were not able to replicate the problem. I already tried to generate wallet with pssphrase disabled, then opened it with empty passphrase successfully. Also reversed (empty passphrase, opened with disabled passphrase Trezor).

I did some tests to confirm my hypothesis, creating 4 wallets with PF-enabled Trezor and 2 wallets with PF-disabled Trezor. I definitely could replicate the issues. 

Steps to reproduce the problem:

1) With PF-enabled Trezor, generated 4 wallets on Monero GUI (A, B, C, D):
- Wallet A (PW blank + PF blank): main address 46...
- Wallet B (PW „abc“ + PF blank): main address 46...
- Wallet C (PW „abc“ + PF „abc“): main address 48...
- Wallet D (PW blank + PF „abc“): main address 48... 

2) Open Trezor Suite and disable passphrase protection

3) With PF-disabled Trezor, generate 2 wallets on Monero GUI (E, F):
- Wallet E (PW blank): main address 46...
- Wallet F (PW „abc“): mains address 46...

4) With PF-disabled Trezor, try opening all wallet files (A, B, C, D, E, F):
- Wallet A (PW blank + PF blank): main address 46... OK
- Wallet B (PW „abc“ + PF blank): main address 46... OK
- Wallet C (PW „abc“ + PF „abc“): main address 48... ERROR: couldn't open wallet. Device wallet does not match wallet address.
- Wallet D (PW blank + PF „abc“): main address 48... ERROR: couldn't open wallet. Device wallet does not match wallet address.
- Wallet E (PW blank): main address 46... OK
- Wallet F (PW „abc“): mains address 46... OK

5) Open Trezor Suite and re-enable passphrase protection

6) With PF-enabled Trezor, all wallets files (A, B, C, D, E, F) open normally
- Wallet A (PW blank + PF blank): main address 46... OK
- Wallet B (PW „abc“ + PF blank): main address 46... OK
- Wallet C (PW „abc“ + PF „abc“): main address 48... OK
- Wallet D (PW blank + PF „abc“): main address 48... OK
- Wallet E (PW blank): main address 46... OK, but now you have to enter a blank PF -> Even though it's working as expected, it's not really intuitive for the user, since when the user created this wallet file his Trezor was PF-disabled, and he didn't have to enter any PF (the user thinks his wallet is PW only, but now, for the 1st time, Monero GUI is asking PW + PF for the same wallet file).
- Wallet F (PW „abc“): mains address 46... OK, but now you have to enter a blank PF

So my final conclusion:

1) If you have a PF-enabled Trezor and create a Monero GUI wallet with a non blank PF, this wallet will stop working if, in the future, you disable PF protection on Trezor Suite. This was the situation observed with user in case 1 (see my comment above)

2) If you have a PF-disabled Trezor and create a Monero GUI wallet, Monero GUI will only ask you a PW. No PF will be asked. The user understand that his Monero GUI wallet is authenticated only by a PW. But, if, in the future, the user enable PF protection on Trezor Suite, now Monero GUI will ask PW + PF for the same file that previously it only asked PW. Now this wallet file will only open if the user types his PW + a blank PF, which is a new situation. Probably the user on case 2 opened his wallet with his PW, but when asked for the PF, he entered his PW again, which resulted in the "couldn't open wallet. Device wallet does not match wallet address." error.

selsta, 2021-07-14:
> I definitely could replicate the issues.

From what I can see the issue is that users claimed that a blank passphrase resulted in a different address compared to passphrase disabled. You don't seem to have replicated that? Your examples show everything working as intended.

ph4r05, 2021-07-14:
Thanks for test coverage @rating89us, this also confirms our hypothesis, that both firmware and monero-wallet-{cli,gui} operate as expected, i.e., disabled passphrase == empty passhprase. 

We could add flag to the wallet file indicating whether passphrase was enabled during wallet creation (PassphraseAck message was received with question PC vs Trezor entry). However, we cannot do this for existing wallets and I am not currently convinced this a bit more complicated feature justifies the purpose. @andrewkozlik ideas?

prusnak, 2021-07-14:
> We could add flag to the wallet file indicating whether passphrase was enabled during wallet creation

Side-note: In the future all Trezor devices will report they have passphrase enabled. And the way how you tell Trezor to "don't use the passphrase" would be by telling it to use the empty passphrase.

That means adding such field does not make such sense.

matejcik, 2021-07-14:
@ph4r05 Monero GUI should actually do the same thing Suite does, and offer a choice between "Standard" and "Hidden" wallet, remember this choice in the wallet file, and ask to enable passphrase if it is disabled and hidden wallet is requested.

We will be moving Connect to the same interface, so this would help with consistency.

Of course it doesn't help existing users, but migration is relatively simple.

rating89us, 2021-07-14:
> see the issue is that users claimed that a blank passphrase resulted in a different address compared to passphrase disabled. You don't seem to have replicated that? Your examples show everything working as intended.

Quoting [reddit topic](https://www.reddit.com/r/monerosupport/comments/obb6uu/issues_accessing_monero_wallet_using_gui_after/) of user in case 1
> Opening from File:
> 1. Open Monero GUI
> 2. Open wallet from file
> 3. Select wallet
> 4. Enter password for wallet
> 5. Error: "Device wallet does not match wallet address"

What happened here was the error reproduced in step 4. Wallets C/D in my tests. The user had a wallet file that was created when he had a PF-enabled Trezor. Then he disabled PF in his Trezor. Now he tries to open again this file in Monero GUI. He entered the correct PW and wallet was decrypted, but since now he is using a PF-disabled Trezor, Trezor returned a blank PF for this non-blank PF wallet file, which resulted in the error "Device wallet does not match wallet address".

selsta, 2021-07-14:
The user wrote multiple times that they don't have any PF set, I think that's where the confusion comes from, see e.g. from Reddit:

> I understand this, and to clarify as far as I know I did not set a passphrase for my Trezor model T, because if I had I would have made a note on a physical copy of my back up seed.

ph4r05, 2021-07-14:
@matejcik 
> @ph4r05 Monero GUI should actually do the same thing Suite does, and offer a choice between "Standard" and "Hidden" wallet, remember this choice in the wallet file, and ask to enable passphrase if it is disabled and hidden wallet is requested.

Ah ok so flag I mentioned would be stored there, just its value will not be derived from Trezor behaviour (as @prusnak pointed out is useless), but on the user choice. 

If user would select "Standard" on passphrase-enabled Trezor, how would that work? Will a) this hoice be somehow reflected in protobuf comm from wallet to Trezor or b) wallet send empty passphrase from PC in that case?

> We will be moving Connect to the same interface, so this would help with consistency.

Will this change protobuf comm protocol? We would need to address this early as these proto changes require some testing. Thanks!

Idea of Standard and Hidden wallets is fine, we could unify it with Suite. But we also need to discuss with monero-devs (e.g., @selsta), on how to implement this device-specific behaviour. Currently, we would have to add these device-specific-options to wallets, so user can specify wallet type during creation, add this to wallet-api and integrate with CLI and GUIs. It's nontrivial effort, so we need to be sure how to approach this. 

Thanks for insight!

rating89us, 2021-07-14:
Actually the user had a PF set for his Monero GUI wallet, see this: 

> As far as I know, I did not have a passphrase set up for my trezor, only the seeds and pin. When access my monero wallet through the GUI I'd enter a password, then enter a passphrase on my trezor device.

matejcik, 2021-07-14:
> If user would select "Standard" on passphrase-enabled Trezor, how would that work? Will a) this choice be somehow reflected in protobuf comm from wallet to Trezor or b) wallet send empty passphrase from PC in that case?

wallet sends empty passphrase.
The protobuf flow now supports this -- you get a `PassphraseRequest` and can respond with `PassphraseAck(passphrase="")` immediately without prompting the user, if you know to do that.

An improvement that can be done in Monero GUI immediately and even without the standard/hidden thing:
when you're opening the wallet, query the empty-passphrase address. If it matches, do not even show the passphrase dialog and proceed straight to wallet.

> Will this change protobuf comm protocol?

no, this is on the UI level only, the protocol changes are already in place

ph4r05, 2021-07-14:
> An improvement that can be done in Monero GUI immediately and even without the standard/hidden thing:
when you're opening the wallet, query the empty-passphrase address. If it matches, do not even show the passphrase dialog and proceed straight to wallet.

@matejcik this is a good idea. However, it is not that straightforward to implement as the passphrase entry is async, via callback notifications to the wallet, so I would have to additional state management that would handle 2 consecutive get_address calls, one with empty passphrase, the other one with user-entered passphrase and bypassing passphrase prompt flow for the first attempt. And only for wallet opening, not creation. So I have to think about this proposal.

matejcik, 2021-07-14:
> So if wallet is opened with empty passphrase, we have to store this somewhere and do not ask passphrase from user anymore during the session

you should keep a Trezor session_id active during the GUI session, in which case you will only be prompted for passphrase once

93497643455437, 2021-07-14:
@andrewkozlik 

>Hi @93497643455437, thanks for helping to diagnose this issue. 

Of course.

>I was under the impression that you did not have a passphrase set up, but after @matejcik's comment and after re-reading your posts, I am in doubt. What does "enter the passphrase" mean? Are you entering an empty passphrase, i.e. just clicking OK, without any input, or are you actually typing in some non-empty passphrase? 

Okay, I can see where the confusion might lie in what I [wrote](https://old.reddit.com/r/monerosupport/comments/obb6uu/issues_accessing_monero_wallet_using_gui_after/h4ua6cl/). There are two passphrases I'm talking about that I should have been more explicit. There is the Trezor model T passphrase (it's empty, no input) and the monero wallet passphrase (non-empty). For step 10 in my post there, I'm talking about the monero wallet's passphrase which the Monero GUI prompts me to enter on the computer or hardware device. I've been instructed to enter it on my hardware device for security reasons, but entering it on the computer or hardware device works equally well.

>While we are at it, does it work the same if you choose entry on the computer vs. if you enter it on the Trezor?

When I'm asked to enter a passphrase (Step 8) for the Trezor model T, it works to enter an empty, no input passphrase on the suite or wallet.

@rating89us

>He wrongly though[t] that he didn't have PF protection enabled on Trezor because when he entered Trezor Suite he just clicked "Standard wallet (no passphrase)" button to access his Bitcoin wallet.

This is partially correct. To hopefully add more clarity, the issue arose because of the discrepancy between how the wallet.trezor.io browser app restores the wallet and how the suite was restoring the wallet. Restoring from the browser you're asked to enable/disable a passphrase at the beginning, the suite automatically enables a passphrase (empty or otherwise) which you would have to later disable. 

I don't remember exactly, but I strongly suspect I initially created my wallet back in March with the Suite, but when I was trying to restore it a couple weeks ago recently I was using the browser and that's where my issue was rooted. I didn't realize the suite had created an empty passphrase for me and when the browser app asked me to enable/disable I selected disable because I had no memory or record of creating a passphrase. Complicating it further, I had no reason to suspect I had a (empty) passphrase hidden wallet because I could see my BTC in the Trezor Suite which led me to believe I had no hidden wallet, but turns out that I did but it contained no BTC. 

Compounding all this, because of my ignorance of my hidden wallet's empty passphrase I suspected it was the shamir backup at the root of my issue, but that was clearly a red herring which is why this issue stated in the original post by ph4r05 no longer applies to my case, but initially it was suspected as a cause for my problem accessing my monero wallet.

I hope this clarifies some things related to my specific issue. Let me know if you want further clarification, I'm happy to help where possible.

rating89us, 2021-07-14:
Thanks for the info, you further confimed my hypothesis:

You had a PF-enabled trezor (created with Trezor Suite), which you used to create a Monero GUI wallet with a non-blank PF. Then you restored your device using Trezor Wallet on browser, which disabled PF protection on your device. After that, you tried to open your old Monero GUI wallet file (which requires a non-blank PF), but since your Trezor was PF-disabled, Trezor returned a blank PF to Monero GUI, which caused the "addresses not matching" error.

93497643455437, 2021-07-14:
Correct!

andrewkozlik, 2021-07-15:
@93497643455437 thanks, this makes your issue perfectly clear.

wendys-cats, 2021-07-15:
Hi, we are taking a look at it again to make sure everytinhg is working as intended (when it comes to Trezor). 
@andrewkozlik is [this](https://github.com/trezor/trezor-firmware/issues/1686#issuecomment-879766277 ) still relevant? And does all of it also apply to 2.4.0? Thank you for answering!

andrewkozlik, 2021-07-15:
> Hi, we are taking a look at it again to make sure everytinhg is working as intended (when it comes to Trezor).
> @andrewkozlik is [this](https://github.com/trezor/trezor-firmware/issues/1686#issuecomment-879766277) still relevant? And does all of it also apply to 2.4.0? Thank you for answering!

@qa  for now please ignore the test cases requested above. Thanks!

andrewkozlik, 2021-07-21:
It appears that @matejcik's [assessment](https://github.com/trezor/trezor-firmware/issues/1686#issuecomment-878288218) is the most likely explanation this issue:

> In summary, so far it seems it's a veritable collection of user errors.

We have tested this quite extensively on various platforms without finding anything unexpected. So to our best knowledge this is not a firmware issue. Much of the confusion seems to have been caused by the passphrase being enabled by default when initializing/recovering Trezor with Suite and disabled by default when doing so with Wallet. I believe the Suite behavior is clearly preferable. I wouldn't create a Wallet issue though, because Wallet is being deprecated in favor of Suite. However, there is an [improvement](https://github.com/trezor/trezor-firmware/issues/1686#issuecomment-879789424) that can be made in Monero GUI for which there already exists an issue https://github.com/monero-project/monero/pull/7793. So as far as I can tell, this can be closed.

@matejcik [asked](https://github.com/trezor/trezor-firmware/issues/1686#issuecomment-878288218) some of the Reddit users for clarification over a week ago without reply. If anything new pops up from them, then we can reopen this issue.

[2 more replies]

closed 2021-07-21 by andrewkozlik