trezor/trezor-firmware #1691: Stellar: add operations

quietbits, 2021-07-07:
Hello Trezor Team,

The following operations are not supported in the Stellar app:
- [Create Claimable Balance](https://developers.stellar.org/docs/start/list-of-operations/#create-claimable-balance)
- [Claim Claimable Balance](https://developers.stellar.org/docs/start/list-of-operations/#claim-claimable-balance)
- [Begin Sponsoring Future Reserves](https://developers.stellar.org/docs/start/list-of-operations/#begin-sponsoring-future-reserves)
- [End Sponsoring Future Reserves](https://developers.stellar.org/docs/start/list-of-operations/#end-sponsoring-future-reserves)
- [Revoke Sponsorship](https://developers.stellar.org/docs/start/list-of-operations/#revoke-sponsorship)
- [Set Trustline Flags](https://developers.stellar.org/docs/start/list-of-operations/#set-trustline-flags)
- [Clawback](https://developers.stellar.org/docs/start/list-of-operations/#clawback)
- [Clawback Claimable Balance](https://developers.stellar.org/docs/start/list-of-operations/#clawback-claimable-balance)

We're adding Trezor support to [Laboratory](https://laboratory.stellar.org/) and it would be nice if all operations could be signed.

JFWooten4, 2021-08-12:
Yeah this is pretty serious. Even setting account clawback flag fails

stepmes, 2021-08-16:
Thank you @quietbits, we're noting this down and will get back to it as soon as we have the capacity to work on this area.

closed 2021-08-16 by stepmes

JFWooten4, 2021-08-16:
@stepmes  Please don't close :( Create claimable balance results in `Failed to execute 'postMessage' on 'Window': function t(e,n){return function(e,t){if(!(e instanceof t))throw new TypeError(\"Cannot call a class as a functio...<omitted>...)} could not be cloned` error, which I don't think is covered by #1756. Can't sing these from Suite to my knowledge

JFWooten4, 2021-10-18:
Any timeline on when this gets fixed? Mission-critical for many use cases

bjorn, 2022-01-19:
It appears claimable balances are getting used more and more. Why not add support for the "Claim Claimable Balance" operation in official firmware? See also the StellarTerm issue linked above.

The current user-experience when trying to sign this transaction is a locked up Trezor One. :-(

yoj6626203, 2022-03-20:
URL: https://stellarterm.com/markets/top/?base=XLM-native <br />User agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/99.0.4844.74 Safari/537.36 Edg/99.0.1150.46 

            Error: Failed to execute 'removeChild' on 'Node': The node to be removed is not a child of this node.
    at Cs (https://stellarterm.com/account/:25:1689294)
    at https://stellarterm.com/account/:25:1708186
    at F.unstable_runWithPriority (https://stellarterm.com/account/:25:2064859)
    at to (https://stellarterm.com/account/:25:1657141)
    at ja (https://stellarterm.com/account/:25:1706006)
    at Sa (https://stellarterm.com/account/:25:1694415)
    at https://stellarterm.com/account/:25:1657420
    at F.unstable_runWithPriority (https://stellarterm.com/account/:25:2064859)
    at to (https://stellarterm.com/account/:25:1657141)
    at io (https://stellarterm.com/account/:25:1657358) 
 
    in div
    in x
    in R
    in div
    in div
    in div
    in div
    in o
    in x
    in R
    in div
    in div
    in div
    in div
    in n
    in e
    in l
    in r