trezor/trezor-firmware #1694: stellar: send does not work again

msab, 2021-07-09:
**Describe the bug**
I am using Model T wallet with Chrome on Mac OS X. I am trying to transfer XLM from my Trezor Model T to Coinbase Pro. Everytime I try and confirm with the model T after entering the information in the model popup, and moving to the model T to confirm, it says "Memo not set" even though I've set the memo. Also, after hitting confirm on the Model T the browser displays a model popup, Screenshot below of error message:

[image]

**Firmware version and revision**
Latest firmware, updated yesterday

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Trezor.io
 - OS: Mac OS
 - Browser Chrome
 - Version 91.0.4472.114 (Official Build) (x86_64)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Trezor.io/start
2. Click on dropdown and select Stellar XLM
3. It transfers me to "Account Viewer," https://accountviewer.stellar.org/dashboard
4. I hit send and type in the required information
5. Confirm on my Model T, I set the memo but it says memo not set
6. I confirm on the Model T to send it
7. Model pops up in browser saying error, screenshot above.
8. Please help

prusnak, 2021-07-09:
Does the send work with the [Exodus wallet](https://www.exodus.com)? It should support Stellar stored in Trezor device.

matejcik, 2021-07-09:
This looks like Account Viewer bug. The memo doesn't seem to be sent to Trezor.

prusnak, 2021-07-09:
@msab please report the issue here: https://github.com/stellar/account-viewer-v2

quietbits, 2021-07-09:
Thanks for reporting this! We're looking into this now. 🤞 
Issue on the Account Viewer: https://github.com/stellar/account-viewer-v2/issues/311

quietbits, 2021-07-09:
@msab Could you please try the Account Viewer now? The fix should be live.

tsusanka, 2021-07-12:
Great, please reopen if needed.

closed 2021-07-12 by tsusanka