trezor/trezor-firmware #1698: Stellar: support muxed accounts

quietbits, 2021-07-09:
Muxed accounts are defined in [CAP-27](https://github.com/stellar/stellar-protocol/blob/master/core/cap-0027.md) and [SEP-23](https://github.com/stellar/stellar-protocol/blob/master/ecosystem/sep-0023.md). Payment to the destination, which is a muxed account, can't be signed with Trezor on [Stellar Account Viewer](https://laboratory.stellar.org/). The same error happens on [Stellar Laboratory](https://laboratory.stellar.org/) as well.

Trezor device shows the correct transaction information on the device, but it fails to sign it:

[image]

matejcik, 2021-07-13:
Muxed accounts are not supported in firmware. This means that if the device shows "correct transaction information," it's
(a) not showing the muxed account at all?
(b) showing account differently -- in which case i don't see how it's showing "correct information"?
(c) displaying the muxed account correctly, and behaving as if it is non-muxed? in which case, muxed account is indistinguishable (on user and protocol level) from non-muxed?

please explain

matejcik, 2021-07-13:
I tried generating a muxed account in Laboratory and sending to it from Account Viewer, but failed:
[image]
So I can't verify the behavior you describe.

stepmes, 2021-08-16:
Thank you @quietbits for this feedback. We're closing the issue now, but we'll keep it in our backlog for future review.

closed 2021-08-16 by stepmes

overcat, 2021-09-13:
> Muxed accounts are not supported in firmware. This means that if the device shows "correct transaction information," it's
> (a) not showing the muxed account at all?
> (b) showing account differently -- [...]

MuxedAccount was introduced in [CAP-27](https://github.com/stellar/stellar-protocol/blob/master/core/cap-0027.md), and [SEP-23](https://github.com/stellar/stellar-protocol/blob/master/ecosystem/sep-0023.md) introduced how to use ASCII to represent this type of address.

The definition of `MuxedAccount` in CAP-27 is as follows:
```
union MuxedAccount switch (CryptoKeyType type) {
 case KEY_TYPE_ED25519:
     uint256 ed25519;
 case KEY_TYPE_MUXED_ED25519:
     struct {
         uint64 id;
         uint256 ed25519;
     } med25519;
};
```

The definition of `AccountID` is as follows:
```
union PublicKey switch (PublicKeyType type)
{
case PUBLIC_KEY_TYPE_ED25519:
    uint256 ed25519;
};
typedef PublicKey AccountID;
```

In Protocol 12, the type of `PaymentOp.destination` is `AccountID`
Currently, Trezor works in this way.
```
struct PaymentOp
{
    AccountID destination; // recipient of the payment
    Asset asset;           // what they end up with
    int64 amount;          // amount they end up with
};
```

In Protocol 13+, the type of `PaymentOp.destination` is `AccountID`
```
struct PaymentOp
{
    MuxedAccount destination; // recipient of the payment
    Asset asset;              // what they end up with
    int64 amount;             // amount they end up with
};
```

Because `MuxedAccount` is partially compatible with `AccountID`, when `MuxedAccount.type` is `KEY_TYPE_ED25519`, the generated XDR data is consistent, so the program can still run well, but when `MuxedAccount.type` is `KEY_TYPE_MUXED_ED25519`, obviously Trezor can't handle it normally.

(a) not showing the muxed account at all?
    When `MuxedAccount.type` is `KEY_TYPE_MUXED_ED25519`, Trezor can't handle it right.

(b) showing account differently -- in which case i don't see how it's showing "correct information"?
    We hope that when `MuxedAccount.type` is `KEY_TYPE_MUXED_ED25519`, M-Strkey can be displayed on the interface.

(c) displaying the muxed account correctly, and behaving as if it is non-muxed? in which case, the muxed account is indistinguishable (on user and protocol level) from non-muxed?
    Sorry, I did not fully understand this question. I think for muxed accounts(when `MuxedAccount.type` is `KEY_TYPE_MUXED_ED25519`), users should be able to see M-Strkey.

In addition, https://laboratory.stellar.org/ works fine now, and you can use it to test Muxed Account.

If you have any other questions, please let me know, thank you.

matejcik, 2021-09-13:
The original report claims that:

> Trezor device shows the correct transaction information on the device, but it fails to sign it:

From what you say, and from my testing, Trezor does not accept a transaction that uses muxed accounts. So "shows correct transaction information" is not true, it does not show _any_ transaction information.

overcat, 2021-09-13:
> The original report claims that:
>
>
> From what you say, and from my testing, Trezor does not accept a transaction that uses muxed accounts. So "shows correct transaction information" is not true, it [...]

Yes, so can we add support for Muxed Account? I would be happy to submit a PR

matejcik, 2021-09-13:
This is currently registered in the backlog. I'll let you know if there's a decision to go forward with implementing more Stellar features. (fwiw we are currently opening a Community Firmware initiative, which will be less restrictive wrt new features)

overcat, 2021-09-13:
Hello @matejcik, thank you for your answer.

Can we possibly see a detailed description of the "Community Firmware" in 2021 Q4?

matejcik, 2021-09-13:
hopefully, but I'm not making any promises.