trezor/trezor-firmware #1699: Stellar: invalid signature if upper time bound not set

quietbits, 2021-07-09:
**Describe the bug**
On [Stellar Laboratory](https://laboratory.stellar.org/) created a payment transaction without the upper time bound. The transaction was shown correctly on the device and signed. When validating the signature on the Laboratory, it was invalid. The signature validates correctly if the upper time bound is set.

**Firmware version and revision**
Trezor One v1.10.1

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Trezor Wallet
 - OS: Windows and MacOS (Trezor Bridge v2.0.27)
 - Browser: Chrome v91.0.4472.124

**To Reproduce**
Steps to reproduce the behavior:
1. Go to [Laboratory](https://laboratory.stellar.org/)
2. Select `public` network (top right corner)
2. Click on "Build Transaction" tab and fill in:
- Source Account
- Transaction Sequence Number (click on Fetch... button)
- Select Payment in Operation Type dropdown
- Destination
- Asset `native`
- Amount
3. Click "Sign in Transaction Signer" at the bottom
4. Scroll to the bottom of the page and click "Sign with Trezor"
5. Follow the Trezor instructions
6. Should see "Success!" message that the signature was added
7. Click on "Submit in Transaction Submitter" at the bottom of the page
8. Scroll to the bottom of the page to view checked signatures

[image]

**Expected behavior**
The upper time bound should not be required to validate the signature.

matejcik, 2021-07-22:
we can't currently verify this as the Laboratory doesn't want to create a transaction where lower bound is set but upper bound isn't.

shanzzam, 2021-07-28:
> we can't currently verify this as the Laboratory doesn't want to create a transaction where lower bound is set but upper bound isn't.

The issue in the Laboratory has been fixed with https://github.com/stellar/laboratory/pull/542

matejcik, 2021-07-29:
cannot reproduce the issue with [this XDR](https://laboratory.stellar.org/#xdr-viewer?input=AAAAAgAAAADjcmgwoLYMtfUshEz%2FzU7tZeulwVXomyZBFWJyTnHlRAAAAGQAAAAAAAAAKgAAAAEAAAAAABVcwAAAAAAAAAAAAAAAAAAAAAEAAAAAAAAAAQAAAADjcmgwoLYMtfUshEz%2FzU7tZeulwVXomyZBFWJyTnHlRAAAAAAAAAAAAJiWgAAAAAAAAAAA&type=TransactionEnvelope)

steps:
1. load Trezor with the following mnemonic: `illness spike retreat truth genius clock brain pass fit cave bargain toe` (SEP-5 test vector)
2. use the above linked XDR for Transaction Signer
3. use the following secret key: `SBGWSG6BTNCKCOB3DIFBGCVMUPQFYPA2G4O34RMTB343OYPXU5DJDVMN`
4. also produce a signature with Trezor

result: both Laboratory and Trezor signature is identical

@quietbits is it possible that you encountered an instance of https://github.com/stellar/laboratory/pull/542 while you were testing this?

quietbits, 2021-07-29:
@matejcik I can reproduce this without any timebounds set on firmware 1.10.2. It's fine with min/max timebounds.

[image]

matejcik, 2021-07-29:
oh that. yeah, we know about that :)

matejcik, 2021-07-29:
I have updated the title and summary to match.

See also https://github.com/stellar/js-stellar-base/issues/438

Currently planned resolution for this issue is to reject "timebounds unset" and only correctly support "timebounds set" -- unless the discussion above goes the other way.

quietbits, 2021-07-29:
Great! Thanks for updating the issue to make it more clear. 👍

PR #1755 (2021-09-24): Strong typing for Stellar
Changes most fields in Stellar protobuf to `required`. This mostly matches the Stellar XDR, so it should not introduce incompatibilities on caller sides.

The notable exception to "mostly matches" are the timebounds fields. In Stellar XDR, they are optional, and in the past, we would treat them as such. This PR makes timebounds required, and so rejects transactions without timebounds.
Per https://github.com/stellar/js-stellar-base/issues/438 omitting timebounds is strongly discouraged by documentation and existing implementations, so I feel that we do not need to supp [... cut]
closed 2021-09-24 by matejcik