trezor/trezor-firmware #1700: Support alpha-numeric "PIN" (a password to unlock the device that is not a BIP39 passphrase)

ecuamobi, 2021-07-09:
**Is your feature request related to a problem? Please describe.**
At the moment it's recommended to use a passphrase so that even if the seed is brute forced the attacker can't access the funds. However if the used passphrase is forgot then there's no way to recover it. Users have to decide whether to use a strong passphrase and risk forgetting it or use a weak one (or have it stored somewhere at hand).

**Describe the solution you'd like**
If an attacker has access to the device then it's possible they can brute force the numeric PIN. However if we can use a complex password instead then brute forcing would be extremely difficult or impossible. At the same time this password would not be a BIP39 passphrase (unless we want it to be) so if we forget it we'd still be able to recover our funds by using the 12- or 24-word backup (or, even better, our Shamir shares).

**Additional context**
It'd be better to enter the password on the device making this compatible with the model T only, unless it's technically possible to enter it on the Trezor One the same way we can enter words during the Advanced recovery process.

prusnak, 2021-07-09:
Since implementing 50-digit PIN (https://github.com/trezor/trezor-firmware/issues/1167) it is possible to use much longer PINs than earlier and this sufficiently satisfies the user story you are describing.

closed 2021-07-09 by prusnak

ecuamobi, 2021-07-09:
According to [random-ize.com](https://random-ize.com/how-long-to-hack-pass) it's more difficult to hack an 8-char password (including uppercase, lowercase, numbers and some special characters) than a 15-digit PIN. It's also easier to memorize.

Higher security would require a 10-character password and is equivalent to a 20-digit PIN, which would be very hard to remember if random.

I insist on this feature even if it has a low priority.