trezor/trezor-firmware #1705: FIDO2 Binance icon missing after origin was changed

bosomt, 2021-07-13:
**Describe the bug**
Biance changed their FIDO2 origin from www.dinance.com to binance.com
As you can see on screenshot there is no icon attached to new origin in current firmware version.

**Firmware version and revision**
2.4.1 regular commit 0x46e05307645e4a446e7c64708f38042aea6e845d

**Desktop/smartphone setup (please complete the following information):**
 - Browser Chrome
 - Version Version 91.0.4472.114 (Official Build) (arm64)

**To Reproduce**
Steps to reproduce the behavior:
1. Open Chrome and login to Binance
2. Go to https://www.binance.com/en/my/security and add new token
3. Observe Binance icon

**Screenshots**

[image]

andrewkozlik, 2021-07-19:
Please download the T1 and TT firmwares from:
https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/pipelines/339348790

For both T1 and TT test the following:

Test case 1:
1. Try to find an old credential that was created when the origin was www.binance.com.
2. Authenticate using the old credential.
3. Verify that the icon shows up correctly (TT only) and that the authentication is successful.

Test case 2:
1. Register a new credential at binance.com.
2. Verify that the icon shows up correctly (TT only), that the origin reads "binance.com", i.e. without the "www", and that the registration is successful.
3. Authenticate using the new credential.
4. Verify that the icon shows up correctly (TT only) and that the authentication is successful.

bosomt, 2021-07-19:
@andrewkozlik ON IT

bosomt, 2021-07-20:
Test case 1:

i was not able to find correct combination of device and account that was registered before they changed origin.
Is it possible to load registration to device via trezorctl or other way ?

Test case 2:

TT **QA OK**
T1  **QA OK**

andrewkozlik, 2021-07-21:
> Test case 1:
> 
> i was not able to find correct combination of device and account that was registered before they changed origin.
> Is it possible to load registration to device via trezorctl or other way ?

These credentials are derived purely from the seed. They do not need to be loaded to the device separately. It should be working fine on our end, so I think we don't need to pursue this.

> Test case 2:
> 
> TT **QA OK**
> T1 **QA OK**

Thanks!

closed 2021-07-21 by andrewkozlik