trezor/trezor-firmware #1708: Error after first swipe in wallet.trezor.io

tsusanka, 2021-07-15:
**Describe the bug**
After initialization of wallet I see error in `wallet.trezor.io` just after I swipe the first word.

**Firmware version and revision**
24bb4016388fca4b998285b95dcd408f4ed0bff6

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Trezor Wallet, not sure about Trezor Suite
 - OS: Linux
 - Browser: Firefox 89

**To Reproduce**
Steps to reproduce the behavior:
1. Wipe device
2. Go to wallet.trezor.io
3. Initialize a wallet
4. Click "Create a backup in 3 minutes"
5. Confirm on device, swipe once and see error in wallet

**Expected behavior**
No error in Wallet.

**Additional context**
**Note that as per @bosomt (@bosomt please confirm here) this happens for emulators only.** So if this is some weirdness in wallet.trezor.io we do not have to worry about it at all and simply close if it is on their side but it felt like curious behavior to me so I have created an issue.

bosomt, 2021-07-15:
@tsusanka i can confirm that i was NOT able to reproduce issue multiple timeswith wallet and 2.4.1 FW during testing before release and after public release

matejcik, 2021-07-15:
can reproduce with emulator from master

I believe it's some sort of udp communication bug. It doesn't seem related to ButtonRequests. In step (5), i can just wait a little while on the swipe screen and Wallet will throw "action was interrupted"

I actually saw something like this in device tests but didn't dive deep enough. I'll do that now.

related: #1668

matejcik, 2021-07-15:
I have a fix! that part is relatively easy.
Let's talk consequences. This does affect hardware, but not too much.

The error in wallet is caused because the emulator stops responding to pings while a paginated screen is up, so the Bridge thinks it's down (see also #1668 again) and sends "device disconnected" event to Wallet.
(It should be possible to reproduce the same error in Wallet by pulling out the USB cable during recovery.)

This does not happen with a real device because we can detect it's still connected, despite not talking to us.

AFAICT, the following doesn't work though.
While on a paginated screen:
# reload Wallet => it will not see the Trezor until you confirm the paginated screen
# issue Cancel event from host. I'm not sure if Wallet or Suite does that, and I believe Connect-based wallets _cannot_ do that. Electrum does it, but then again Electrum doesn't see any paginated screens

matejcik, 2021-07-19:
QA scenarios when fixed:

- what @tsusanka said above, using _emulator_:
	1. Wipe device
	2. Go to wallet.trezor.io
	3. Initialize a wallet
	4. Click "Create a backup in 3 minutes"
	5. Confirm on device, swipe once and see error in wallet
- same but in Suite
- using _hardware device_:
	1. go through steps 1-5 above
	2. close Wallet
	3. reopen Wallet
	4. ensure that everything works fine (failure case: the Trezor reports as "disconnected" and does not respond)
- same but in Suite

matejcik, 2021-07-20:
@bosomt can I trouble you to check this with builds from [here](https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/pipelines/339830693)?

bosomt, 2021-07-20:
Sure :)

bosomt, 2021-07-20:
Tested with _hardware device_

Wallet - waited after one swipe and then proceeded with backup
Suite - waited after one swipe and then proceeded with backup

Suite and wallet with interrupted backup via close or reload of webpage - ended with **Backup failed** as expected, no error on device display just  BACKUP FAILED! message

I'm not able to start Trezor emulator or Trezor user env on my Linux computer, but rest of QA will have more luck hopefully.

wendys-cats, 2021-07-21:
**Tested with emu and with HW device** 

**Error -** displayed during backup (after first swipe) in Wallet/Suite about failed backup due to device being disconnected, no error message displayed on device except standard BACKUP FAILED!

**Wallet + emu - error triggerd**
trezor-user-env master setup: 2.4.1 FW, 2.0.31 bridge. Tested on wallet.trezor.io

**Suite + emu - error triggered without closing Suite**
trezor-user-env master setup: 2.4.1 FW, 2.0.27 bridge. Tested on suite.trezor.io
[image]

**Suite + HW - error triggered with closing  Suite**
 2.3.6 FW, webusb, Chromium 88. Tested on suite.trezor.io

matejcik, 2021-07-21:
@wendys-cats what is "trezor-user-env master"? do you know commit id that you used?

tsusanka, 2021-07-21:
@wendys-cats you can use `trezorctl get-featuers` to find out (field `revision`).

wendys-cats, 2021-07-21:
@matejcik @tsusanka I will check right away, but I pulled and ran the needed .sh before trying out.

tsusanka, 2021-07-21:
@wendys-cats it is a commit hash, just with `0x` in front of it :). That [revision](https://github.com/trezor/trezor-firmware/commit/24bb4016388fca4b998285b95dcd408f4ed0bff6) is from Jul 13th, so something is definitely wrong in trezor-user-env. I have created an issue and @grdddj will sure investigate.

tsusanka, 2021-07-22:
@wendys-cats I receive the correct commit now. Please try again and follow [these steps](https://github.com/trezor/trezor-user-env/blob/b73c04ae5ce861013065e1f6321a1521b9ae2dfc/README.md#run-it) exactly. Mainly the _pull_ part (3rd point) is imporatant. You should get `10d72b964fd71840595b25ea35a147c73b468fac`.

sorooris, 2021-07-23:
It is crucial to select the correct firmware version when running the emulator, and it was not at all obvious to us in QA that the 2-master should be selected - [big thanks to Jiri](https://github.com/trezor/trezor-user-env/issues/91#issuecomment-885633083) for explaining this. Somehow I expected there to be a newer minor version - which wasn't present - instead of "2-master", so I picked the next best thing, 2.4.1, and did not think of scrolling all the way down.

[image]

```
$ trezorctl get-features | grep revision
    revision: 20 bytes 0x8da978981eaa3eb7577319171dc98c88975edd4a
```

STew790, 2021-07-23:
**QA OK**

**Wallet & emulator (8da978981eaa3eb7577319171dc98c88975edd4a)**

Seed created without any errors

**Suite and emulator (8da978981eaa3eb7577319171dc98c88975edd4a)**

The seed was also created without any errors.

matejcik, 2021-07-23:
thank you all! closing as fixed :)

closed 2021-07-23 by matejcik

tsusanka, 2021-07-23:
Btw I think in https://github.com/trezor/trezor-firmware/issues/1708#issuecomment-884281589 I was confused myself and was launching 2.4.1. We will move the `x-master` as the first option in trezor-user-env, so it is more visible https://github.com/trezor/trezor-user-env/pull/96.

PR #1709 (2021-07-20): fix(core): properly respond to USB events while on a paginated screen
fixes #1708