trezor/trezor-firmware #1713: I fail to reproduce firmware trezor-1.9.4.bin

Giszmo, 2021-07-16:
**Describe the bug**

I fail to account for all bytes in trezor-1.9.4.bin

**Firmware version and revision**

1.9.4

**Expected behavior**

The website should prominently feature a procedure to verify all bytes that cannot be reproduced.

**Additional context**

In an attempt to account for all bytes in the file `trezor-1.9.4.bin`, I find 256 + 192 bytes that I cannot check. The latter is probably three signatures but the former I could not even find what it's supposed to be.

How can I with linux standard tools extract and check those signatures? Which public key should they match to? What are the first 256B?

My full review can be found on [WalletScrutiny](https://walletscrutiny.com/hardware/trezorOne/).

prusnak, 2021-07-16:
High-level comment:

Current firmwares are distributed as wrapped images where you have a firmware signed with new method (using https://github.com/trezor/trezor-firmware/blob/master/legacy/bootloader/firmware_sign.py) and the result is then signed using the old method (using the method predating commit https://github.com/trezor/trezor-firmware/commit/07231d936e41335b3ec44c4c6eb336be006890d0).

In both signing operations the keys used are private keys belonging to these public keys: https://github.com/trezor/trezor-firmware/blob/3d69ca1b1b0c0e429474321ced0d040d1d9e6dec/legacy/bootloader/signatures.c#L36

We do the wrap thing so the old firmwares not understanding the new firmware header can apply the firmware update properly. For new firmwares, the Suite and trezorctl just strip the first 256 bytes and use the new image only.

In the near future we will only be providing firmwares only using the new format (and older devices would need to go through a intermediary step of updating the firmware twice).

matejcik, 2021-07-16:
as specified [in a comment](https://github.com/trezor/trezor-mcu/issues/459#issuecomment-470287085) in the issue you link in the review, the header at start of the downloaded image is:
- 256 bytes of the old TRZR header -- which is [stripped](https://github.com/trezor/trezor-firmware/blob/master/python/src/trezorlib/cli/firmware.py#L335-L351) when you upload the image into bootloader 1.8.0 or newer
- 1024 bytes of the new TRZF header, which should be identical to what you built, except for the 192 bytes of signatures

Structure of the old header is described here:
https://github.com/trezor/trezor-firmware/blob/master/python/src/trezorlib/firmware.py#L246-L260
Structure of the new header is described here:
https://github.com/trezor/trezor-firmware/blob/master/python/src/trezorlib/firmware.py#L181-L213

Signature verification for the old header is implemented here:
https://github.com/trezor/trezor-firmware/blob/master/python/src/trezorlib/firmware.py#L411-L418
Namely:
1. calculate digest of firmware without header. This is essentially `tail -c +257 trezor.bin | sha256sum`
2. the `key_indexes` will tell you which 3 of the 5 [`V1_BOOTLOADER_KEYS`](https://github.com/trezor/trezor-firmware/blob/master/python/src/trezorlib/firmware.py#L33-L42) are used
3. the `signatures` are ecdsa signatures of the digest with the respective keys

Signature verification for the new header is identical, except:
- signatures are over a "header digest", which is a sha256 over the header with signature fields zeroed out
- key indexes are `v1_key_indexes` in the new header
- signatures are `v1_signatures` in the new header.

For purposes of reproducibility, we should also publish instructions along the same lines as for TT, i.e., use `dd` to zero out signatures, compare files byte-for-byte.

Giszmo, 2021-07-16:
Thanks for the quick reply. I hold Trezor in highest regards but think I owe it to readers of walletscrutiny, to be rigorous in the process every time and the above information is not straight forward. "zero out what's not reproducible" is not a solution so I will have to make sure whatever is loaded on the device is reproducible or actual valid signatures but the pointers above are probably enough.

prusnak, 2021-07-16:
> but the pointers above are probably enough.

I am closing the issue in this case. Thanks

closed 2021-07-16 by prusnak