trezor/trezor-firmware #1714: Be stricter about downloading the Alpine container

matejcik, 2021-07-16:
https://walletscrutiny.com/hardware/trezorOne/ points out that we just download a container from Alpine CDN, without even using https.

We should at least be checking its hash against a known value.

matejcik, 2021-07-16:
@vdovhanych maybe you could look into this?

matejcik, 2021-07-16:
the download happens in `environment.yml` and in `build-docker.sh`

vdovhanych, 2021-07-16:
Well, we do not download a container from the Alpine CDN. We download Alpine minirootfs itself and create the container with it as a base. I think that it's done this way as there is no minirootfs version on the docker hub. We should change it to https, though. Or another option would be to use the full Alpine image and get it from the hub directly in `Dockerfile` instead of `FROM scratch`, use `FROM alpine:3.14.0` and build it. `build-docker.sh` isn't used anywhere in the ci I assume it for local building but that should also be changed to https.

prusnak, 2021-07-16:
Let's just change to https and check the sha256 hash of the downloaded file.

PR #1726 (2021-07-20): feat(ci): add sha checksum for alpine download and change to https
This should add sha checksum and changes the download of alpine to https. For our GitLab ci and also inside the 
`build-docker.sh ` script. Both should fail with a message `sha256sum: WARNING: 1 computed checksum did NOT match` script and ci will fail if the sha of the file does not match the one in the script. 
Resolves #1714
closed 2021-07-20 by tsusanka