trezor/trezor-firmware #1715: trezorctl firmware download doesn't work if your Trezor is not connected

matejcik, 2021-07-16:
The `firmware download` subcommand requires Trezor to be plugged in. This is impractical.
Moreover, even if I specify an exact version via `-v 1.9.4`, if I have a TT plugged in, the version is not found because it's looking for TT versions only.

Implementation-wise, `download` should only be using `find_best_firmware_version` if version is not specified. Otherwise it should figure out model from the major version, fetch the appropriate releases.json, verify that the version exists and then download it.

Not sure how much functionality we can share between `find_best_firmware_version` and this new thing (`find_specified_firmware_version` maybe?)

@grdddj want to look at this one?

grdddj, 2021-07-16:
Hello, yes, I think I should take this one, as it is connected with my changes.

If it is not a critical thing, I would try it next week, there are still some trezor-user-env issues to be resolved.

matejcik, 2021-07-16:
definitely not critical.

matejcik, 2021-07-21:
fixed in #1721

closed 2021-07-21 by matejcik