trezor/trezor-firmware #1720: Enable to recognize device with intermediary from device with BL 1.10.0 and FW already installed

matejkriz, 2021-07-19:
**Is your feature request related to a problem? Please describe.**
I have T1 with intermediary installed (so no FW, only BL) and I have no chance to finish the installation of latest FW using Suite - because Suite can't recognize the current state of the device from the one with current FW installed but connected in bootloader mode (see the list of features below).

_How did I get into this state in the first place_? During firmware update from FW 1.6.0, BL 1.3.1 after intermediary installation the device wasn't restarted automatically but ask me to reconnect the device (I can't duplicate that, have no more devices with such an old BL.) But probably there are other possible ways to get to that state (disconnect during update or failure of latest FW installation)?

**Describe the solution you'd like**
Would it be possible to change the `firmware_present` value for intermediary on `false`? Or add any other flag to be able to recognize intermediary? (Maybe it's connected to [this](https://github.com/trezor/trezor-firmware/issues/1674#issuecomment-864586447) somehow?

**Describe alternatives you've considered**
We should add in Suite for that case option to install firmware, but possibility to recognize the intermediary would help even in that case for better copy because we can't know if there is already some FW installed and what version it is.

**Additional context**

How does it look in Suite now?
[image]
(Reconnecting the device will not help here, because the device always starts in BL mode...)

Here is the device features of the Trezor with intermediary installed and also features of device with latest firmware installed, just in bootloader mode (the features are the same):
```json
{
    "vendor": "trezor.io",
    "major_version": 1,
    "minor_version": 10,
    "patch_version": 0,
    "bootloader_mode": true,
    "device_id": null,
    "pin_protection": null,
    "passphrase_protection": null,
    "language": null,
    "label": null,
    "initialized": null,
    "revision": null,
    "bootloader_hash": null,
    "imported": null,
    "unlocked": true,
    "firmware_present": true,
    "needs_backup": null,
    "flags": null,
    "model": "1",
    "fw_major": null,
    "fw_minor": null,
    "fw_patch": null,
    "fw_vendor": null,
    "fw_vendor_keys": null,
    "unfinished_backup": null,
    "no_backup": null,
    "recovery_mode": null,
    "capabilities": [
        "Capability_Bitcoin",
        "Capability_Bitcoin_like",
        "Capability_Crypto",
        "Capability_Ethereum",
        "Capability_Lisk",
        "Capability_NEM",
        "Capability_Stellar",
        "Capability_U2F"
    ],
    "backup_type": null,
    "sd_card_present": null,
    "sd_protection": null,
    "wipe_code_protection": null,
    "session_id": null,
    "passphrase_always_on_device": null,
    "safety_checks": null,
    "auto_lock_delay_ms": null,
    "display_rotation": null,
    "experimental_features": null
}
```

matejcik, 2021-07-20:
Intermediary fw actually runs the same bootloader that's in regular firmware. Modifying that is extremely fraught (we're running out of available space for code ATM). So we'd like to avoid that if at all possible.

It seems that adding option in Suite to upgrade firmware at this stage (plus perhaps factory reset?) would be best.
If there are additional concerns, it will be probably best to discuss it on a call with @prusnak.

tsusanka, 2021-07-20:
Can't we solve this by simply versioning the Intermediary FW somehow exclusively, e.g. `9.9.0`? It seems to me to have two firmwares with versions 1.10.0 not a good idea anyway.

prusnak, 2021-07-20:
But the version from Features in the OP is from the bootloader - and this bootloader is indeed of version `1.10.0`

tsusanka, 2021-07-21:
I am not sure this is indeed the issue. Suppose Suite (or whoever, let's say _host_) is installing the Intermediary FW. Once the intermediary FW is installed the user unplugs the device for whatever reason and plugs it back in. If I am not mistaken the intermediary deletes itself after the installation, therefore the device boots up in the bootloader once it is connected to the host again. Host now sees a device in BL mode with `firmware_present = True` (I want to double check this but that is what @matejkriz is saying right?), so the client can't distinguish this case with the one where the user has regular 1.10.0 installed. 

If the fact `firmware_present` is `true` is indeed correct I think that is a clear bug as the intermediary deletes itself afterwards nope? Where am I wrong?

cc @hiviah

prusnak, 2021-07-21:
> If the fact `firmware_present` is `true` is indeed correct I think that is a clear bug as the intermediary deletes itself afterwards so it does not have a firmware per-se. No? Where am I wrong?

Well, the device has the firmware, but it is deactivated. If it did not have the firmware, the storage would be deleted.

prusnak, 2021-07-21:
> so the client can't distinguish this case with the one where the user has regular 1.10.0 installed.

Client can NEVER distinguish what firmware version is installed, because bootloader on T1 does not report this. The reported version (1.10.0) is the version of the bootloader.

Clearly, Suite is making some assumptions and I think that is not right. If a device is connected in bootloader mode, there should be a firmware update offered to the last version.

tsusanka, 2021-07-21:
> Client can NEVER distinguish what firmware version is installed, because bootloader on T1 does not report this. The reported version (1.10.0) is the version of the bootloader.

I just wanted to write this. So versioning the intermediary as `9.9.9` (or whatever) does not solve anything.

tsusanka, 2021-08-10:
As discussed we will solve this in Suite. @matejkriz will create an issue in trezor-suite.

closed 2021-08-10 by tsusanka