trezor/trezor-firmware #1734: Unable to click the Continue button by press_yes function

grdddj, 2021-07-27:
**Describe the bug**
Automatically pressing `Continue` button at the end of recovery process by using `press_yes` debuglink function is not successful, even though `press_yes` works in all other cases when confirming something

**Firmware version and revision**
Firmware versions higher than 2.1.4 (there it is functional) - tested on 2.3.1 and 2.4.0.

**To Reproduce**
Steps to reproduce the behavior:
1. Running Suite e2e tests - specifically [t2-recovery-success.test.ts](https://github.com/trezor/trezor-suite/blob/e1c710e0c6bbd96b0e7970c15e3650c092aa0c83/packages/integration-tests/projects/suite-web/tests/onboarding/t2-recovery-success.test.ts#L12) and choosing some newer version of firmware (other than specified 2.1.4)
2. The test fails as [this press_yes](https://github.com/trezor/trezor-suite/blob/e1c710e0c6bbd96b0e7970c15e3650c092aa0c83/packages/integration-tests/projects/suite-web/tests/onboarding/t2-recovery-success.test.ts#L38) does not do anything on the emulator

**Expected behavior**
`press_yes` function clicks the `Continue` button

**Screenshots**
[image]

**Additional context**
Connected with Suite e2e tests and trezor-user-env:
- [Suite issue](https://github.com/trezor/trezor-suite/issues/4069)
- [Tenv PR](https://github.com/trezor/trezor-user-env/pull/97)

prusnak, 2021-07-27:
> even though `press_yes` works in all other cases when confirming something

Is that really the case? There are two different types of buttons in the UI - Confirm and HoldToConfirm. Does `press_yes` works for other instances of HoldToConfirm?

grdddj, 2021-07-27:
In suite e2e tests, it is being used literally everywhere to confirm things on emulator, I did not see any other "confirmation" function. It also speaks to `emulator` through `trezor-user-env`, and `press_yes` is the only command we expose

matejcik, 2021-07-27:
`press_yes` should generally work on both Confirm and HoldToConfirm

prusnak, 2021-07-27:
> `press_yes` should generally work on both Confirm and HoldToConfirm

Thanks for clarification

matejcik, 2021-07-27:
per [our device tests](https://github.com/trezor/trezor-firmware/blob/matejcik/layouts-4/tests/device_tests/test_msg_recoverydevice_bip39_t2.py#L62-L63), this button works fine for `press_yes`.

@grdddj can you provide emulator logs matching the testcase?

matejcik, 2021-07-27:
please also try this with emulator from current master, which tells us more about received messages

grdddj, 2021-07-27:
> @grdddj can you provide emulator logs matching the testcase?

When tried locally on `master`-built emulator, it is fine (logs below). I will try to get all the emulator logs from the running test where it fails to click.

It could be something with Suite tests. I see that @mroz22 made [these comments](https://github.com/trezor/trezor-suite/blame/e1c710e0c6bbd96b0e7970c15e3650c092aa0c83/packages/integration-tests/projects/suite-web/tests/onboarding/t2-recovery-success.test.ts#L12-L13) more than a year ago - and it still applies. 

[image]

matejcik, 2021-07-27:
for the record, the above is probably not current `master`, as it does not include commit cb882df100dc33f753650e1aeecc126c05246340

otherwise you'd see something like this:
[image]

grdddj, 2021-07-28:
> for the record, the above is probably not current `master`, as it does not include commit [cb882df](https://github.com/trezor/trezor-firmware/commit/cb882df100dc33f753650e1aeecc126c05246340)

You were right, my local emulator binaries were not updated.

The log below is all the output we got from that [Suite test](https://github.com/trezor/trezor-suite/blob/e1c710e0c6bbd96b0e7970c15e3650c092aa0c83/packages/integration-tests/projects/suite-web/tests/onboarding/t2-recovery-success.test.ts). The `2-master` emulator is being started at line 610. It still did not click the `Continue` button.

Test context: latest image of `trezor-user-env`, `develop` branch of Suite, running the test locally using containers via `docker/docker-suite-test.sh` Suite command

[t2-recovery-success-test.log](https://github.com/trezor/trezor-firmware/files/6891608/t2-recovery-success-test.log)

matejcik, 2021-07-28:
You're inputting `"all"` 13 times, so in particular you're sending `"all"` input to the last screen. That will not magically work and the last click does not go through.

closed 2021-07-28 by matejcik

grdddj, 2021-07-28:
Nice catch, thanks for noticing! After sending it only 12-times, everything is fine.