trezor/trezor-firmware #1737: PIN/Passphrase input doesn't work in some environments

vietnamese-chowder, 2021-07-27:
**Describe the bug**
I'm trying to run trezorctl to sign a Qtum transaction built in Java alongside a Trezor device, by creating a process through ProcessBuilder that runs the command `trezorctl sign-tx <name_of_file>.json`. After running it, I'm feeding it the PIN (corresponding to the layout) that I acquire from the user on the frontend.

I am inputting the PIN to the process using a BufferedWriter(OutputStreamWriter(proc.getOutputStream())) by calling write(pin), then newLine(), then flush(). Everything works on Linux and on Mac when running the program through Intellij, but when I build the jar file and run that, it doesn't. trezorctl doesn't receive the input and my program hangs at the next stdInput.next(). On Windows this behaviour also occurs on Intellij, and on Git Bash.

When I run the command line application manually on Windows, it works on cmd, but not on Git Bash, exhibiting the same behaviour. I've also tried running my program with sudo, so I don't think it's a permission issue. I'm pretty sure this is an issue with trezorctl's IO and not my program, because I tested the exact same method on a dummy Python script I wrote and it worked, even on Windows.

**Firmware version and revision**
1.10.2

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: trezorctl
 - OS: Windows
 - Browser: None, Git Bash
 - Version: 0.12.2

**To Reproduce**
Steps to reproduce the behavior:
1. Make sure your Trezor has either PIN or Passphrase enabled
2. Open Git Bash on Windows
2. Type any trezorctl command i.e. `trezorctl get-address -n "m/49'/0'/0'/0/0"`
3. Try typing your PIN/Passphrase (according to the layout) and press enter

**Expected behavior**
The mask doesn't work, so you'll see what you're typing, and pressing enter does nothing.

**Screenshots**
[image]

matejcik, 2021-07-28:
This seems to be an issue either with the click library, or with Python itself.

possibly related: #1682 

please try manually upgrading click: `pip install --upgrade click` and see if the problem goes away afterwards

matejcik, 2021-07-28:
The issue is actually a [bug](https://bugs.python.org/issue44762) in Python's `getpass` function, which does not work correctly on Windows unless you're using a real terminal.

(Weirdly enough, Git Bash is using MinTTY which is not actually a real terminal. It can be fixed by enabling `winpty` while installing Git for Windows.)

I suppose we could explicitly detect the condition and fall back to reading non-hidden input.

I'm also wondering if we should do something more advanced, e.g., like HWI does, for separating PIN entry and subsequent actions, for usage in scripts.

matejcik, 2022-01-12:
@vietnamese-chowder please have a look at the new `--script` option to `trezorctl` for your usecase

closed 2022-01-12 by matejcik