trezor/trezor-firmware #1746: Unknown bootloader detected

btclinux, 2021-08-02:
Hi! I built firmware with docker script, flash the bootloader and install firmware with trezor suite. So, I have "Unknown bootloader detected" error on device screen. And my question why it happens? Also I unable to connect to device with programmer anymore what I asked for in hardware repository https://github.com/trezor/trezor-hardware/issues/18

btclinux, 2021-08-02:
Finally I went to bootloader mode and can to reflash the firmware that I built with docker script. But after installing now I have "Hard fault detected" on device screen. And I still can't to connect to device via programmer. How to erase STM now or reflash the bootloader?

prusnak, 2021-08-02:
Please read the instructions here https://github.com/trezor/trezor-firmware/blob/master/docs/legacy/index.md#how-to-install-custom-built-firmware

closed 2021-08-02 by prusnak