trezor/trezor-firmware #1747: Missing message for Failure_ActionCancelled on T1

marekrjpolak, 2021-08-04:
**Describe the bug**
When updating firmware via Suite but aborting the update on device (T1), device doesn't return error message:
`FirmwareErase -> { code: "Failure_ActionCancelled", message: null }`
([probable source](https://github.com/trezor/trezor-firmware/blob/0e650b1a5be5c74a58394d35be1dba3d356a5053/legacy/bootloader/usb.c#L247))

For comparison, when wiping device and aborting it in the same manner, device returns correct message:
`ButtonAck -> { code: "Failure_ActionCancelled", message: "Action cancelled by user" }`
([probable source](https://github.com/trezor/trezor-firmware/blob/0e650b1a5be5c74a58394d35be1dba3d356a5053/legacy/bootloader/usb.c#L210))

**Firmware version and revision**
Tested on 1.10.1.

**Desktop/smartphone setup**
Current web Suite.

**Additional context**
Possibly related to #1334.

matejcik, 2021-08-04:
you should not rely on contents of `message`

closed 2021-08-04 by matejcik

matejcik, 2021-08-04:
Additionally: we are SEVERELY space-constrained in T1 bootloader, enough so that adding a string message for this would actually be a problem. So even if there's compelling reason on the Suite side, this is close to impossible on the bootloader side.

But in any case you should not be relying on contents of `message` field.