trezor/trezor-firmware #1749: Support Ledger Live dereviation

mastercyb, 2021-08-05:
* I'm not sure that this repo is the correct location of this feature request.

I've used Ledger for several years. Now I am pretty sure I am not gonna use the product of this company ever. Reasons are:
- no respect to the user
- no privacy guaranties
- VPN dependency
- almost zero reliability of the device
- low usability of the device given recent changes

The only question is how to switch from this hell. While seed phrases are interoperable, derivation paths are not. A does not have an option to transfer to the different account because (1) it would be crazy expensive and (2) in several cases I need several years due to different locks. 

So if Trezor want support from a horde of guys who was fucked by Ledger, I beg you - add support for their derivation path

matejcik, 2021-08-05:
You'll need to be more specific than that. There is no One True Derivation Path. Each coin can use one or more derivation paths. As far as we know, Bitcoin and bitcoin-like derivation paths are standardized, and the same between Ledger and Trezor.

stepmes, 2021-08-16:
Sorry to hear about your bad experience. It also seems to me that you should be able to switch by simply performing a wallet recovery on Trezor using your existing backup seed. If you have a more specific use case, please feel free to contact our support team at https://trezor.io/support/ they would be happy to help.

matejcik, 2022-07-06:
@saurik let me repeat myself:

> You'll need to be more specific than that. There is no One True Derivation Path. Each coin can use one or more derivation paths.

saurik, 2022-07-07:
@matejcik I provided the exact derivation path for Ledger Live, so I'm not sure how much more specific I can be.

psypher246, 2022-08-10:
Hello,
@saurik is correct. And there are many articles/posts about how and why Ledger Live does this and why this breaks things for people moving their wallets to other devices. I've spent 3 days down the rabbit hole to gain access to my funds again.

Ledger Live seems to create all the Ethereum Wallet under an HD path of m/44'/60'/0'.

So when you have migrated/restored your passphrase from Ledger to Trezor, Trezor Suite does not detect your Ledger Live as the default path, instead the default path for Trezor/MEW/Metamask etc (m/44'/60'/0'/0), which if you were using Ledger Live, now shows zero funds in any wallets and none of the address are familiar.

When you use Metamask with Ledger it asks you which HD path do you want to use: Ledger Live Legacy (m/44'/60'/0')  or another (sorry I can't remember right now what it was called, maybe "default", my Ledger is broken hence migration to Trezor) The other path is basically the default path for other wallets which is the one Trezor finds with no funds.

So as per @saurik, in order to access your funds you can ONLY use MEW and you must choose Lightstreams Network (m/44'/60'/0'), (Metamask does not detect the other HD path from the trezor when it launches and you can only choose the default one) and once you use that path you can see all your Ledger Live funds. But then when you try send funds you get the error: "Not a valid path"

I found a fix for this however based on this post: https://forum.trezor.io/t/how-to-fix-error-forbidden-key-path/8642 
If you set the safety checks to "Prompt" then the trezor no longer sends the error to MEW stating this is an invalid path and you can send your funds.

So Ideally, Trezor and Trezor Suite should be aware of this use case and when discovering HD paths it should also discover this Ledger Live HD path (for apps like Metamask) and give the user an option or mechanism to select their default path in some way.

psypher246, 2022-08-10:
Additionally, when you import your trezor into brave wallet, it only shows a "Default" HD path with an option to switch but the drop down list only shows Default, which has zero funds. When you do the same with Ledger, it shows both the Legacy Ledger Live Path and the Default one. Choosing Ledger Live shows all your funds. So whatever mechanism Ledger Live is using to list those HD paths in these Web Wallets, is what's missing from Trezor, and without this, no Ledger Live user can easily migrate to Trezor. See attached screenshot. 

[image]

matejcik, 2022-08-10:
> So as per @saurik, in order to access your funds you can ONLY use MEW and you must choose Lightstreams Network (m/44'/60'/0'), (Metamask does not detect the other HD path from the trezor when it [...]

Thank you, this is the first piece of information that actually helps with reproducing the issue.

I am reopening the issue an updating the summary.

> 

> when you import your trezor into brave wallet, it only shows a "Default" HD path

this is unfortunately fully on Brave. Trezor has no mechanism for indicating the HD paths to wallets -- so ISTM Brave is detecting Ledger specifically and adding the option.

psypher246, 2022-08-10:
@matejcik thanks for confirming. I guess we would need to reach out to Brave and Metamask to support this and for Trezor not to see it as a security risk so the prompt setting is no longer needed. 

Is there any way to get Trezor suite to interact with this HD path?

closed 2022-11-30 by marnova

bosomt, 2022-12-13:
**QA OK**

_2.5.3_
```
trezorctl ethereum get-address --address "m/44'/60'/0'/0"
Error: DataError: Forbidden key path
```
_2.5.4_     revision: ff35c77bad82c7acac1df6245940ebe0d6454642
```
trezorctl ethereum get-address --address "m/44'/60'/0'/0"
0x1ec5C1854e3E9F1674c34D6C2Be1bf13DFc0Fd8F
```

kanzure, 2023-05-05:
Why not just let users feed in an arbitrary bip32 path? It's just bip32. "m" is a valid bip32 path. If valid bip32 paths aren't going to be accepted then why even call it bip32?

kanzure, 2023-05-09:
My use case is taking the master xpub and deriving different addresses for the user to pick from until they see the one they recognize. If I start with "m", I can run through all the different address types. But if I have to do a roundtrip to the hardware wallet for each account/address type in the bip44 path then this would be a poor user experience. (Why would a user remember if it's p2wpkh or p2pkh etc? They might not even know the difference.)

FWIW I do not get "Forbidden key path".

I recognize this is probably off-topic for this issue so I'll open a different issue somewhere else if this persists. "Not a valid path" appears to be from this layer:
https://github.com/trezor/connect/blob/develop/src/js/utils/pathUtils.js