trezor/trezor-firmware #1751: Do not convert bech32 addresses to upper-case for QR code

matejcik, 2021-08-06:
Per https://www.qrcode.com/en/about/version.html, V3 QR code can fit 42 characters in BINARY mode, and V4 QR code can fit 62. This happens to match the length of Bitcoin segwit bech32 addresses for single-sig (42) and multisig (62) accounts.

Despite that, we convert bech32 addresses to uppercase. Ostensibly to lower complexity of the QR code, but this does not help in practice (V2 fits only 38 ALPHANUM so we need V3 anyway; conversely, V3 fits only 61 ALPHANUM so we need V4 anyway).

We should probably keep bech32 addresses as lowercase, for better compatibility.

How are the lengths of taproot addresses?

prusnak, 2021-08-07:
> How are the lengths of taproot addresses?

The same as segwit v0 multisig. Here is a valid segwit v1 taproot address: `bc1pw508d6qejxtdg4y5r3zarvary0c5xw7kw508d6qejxtdg4y5r3zarvary0c5xw7kt5nd6y`

matejcik, 2022-03-10:
QA instructions:

1. show QR codes for Segwit and Taproot addresses on device, scan them, ensure that the contents match the address and that the scanned text is in lower case.
2. show QR codes for different kinds of addresses (incl. other currencies than Bitcoin), ensure the QR codes can be scanned and are in the expected format

bosomt, 2022-03-11:
On it
@matejcik just model T right ? i don't see any change of code in model One code

matejcik, 2022-03-11:
exactly. For step 2 you can compare results to T1 if it's unclear what "expected format" is

bosomt, 2022-03-11:
found one issue compared to 2.4.3
we lost BITCOINCASH: prefix in QR code 
so you will get different results on device display and different results displayed in suite
+ Suite can match this address to correct account 

[image]

2.4.3: `BITCOINCASH:QPQ7CUGZRVNMKGFS3TG0H5YRF5M6NLNACYWR7UTCQT`
2.4.4: `qry3epyn9ulff5mnz7d89j3ks30acvmx5gx3tm3f62` displayed QR code and address

matejcik, 2022-03-11:
is there a difference between previous a current FW?

matejcik, 2022-03-11:
In case of vtc (and possible other bech32 currencies), it's another question -- as Bitcoin bech32 addresses are 42 characters but VTC is 43 -- so the QR code would actually need to be smaller (higher-res) to fit the lowercase encoding.

about BCH, please re-test with https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/pipelines/489913251

matejcik, 2022-03-11:
Reopening the issue as the original description still applies for the VTC problem.

A Product decision is needed (cc @sime): how much do we care about (a) QR code legibility, which _may_ already be too low to be of concern, and (b) functioning of coins like VTC with regard to QR code format.

To add: while bech32 addresses are not case-sensitive, apparently some wallet software will reject upper-cased address variants.

bosomt, 2022-03-11:
revision: 20 bytes 0x0bfdfbe925f64890619325cd0cc7f5c296f491da,

**VTC**: same behavior as 2.3.4
**BCH**: BITCOINCASH: prefix and UPPER CASE is back, but suite says invalid address

[image]

matejcik, 2022-03-11:
> BITCOINCASH: prefix and UPPER CASE is back, but suite says invalid address

did Suite previously accept this and now doesn't?

bosomt, 2022-03-11:
yes :) and when i remove prefix it accepts it ;)

[image]

matejcik, 2022-03-11:
that's kind of weird.

To be extra clear: what is the contents of the QR code on 2.4.3 ? Does Suite accept it?
What is the contents of the QR code on T1? Does Suite accept it?

bosomt, 2022-03-11:
so QR code for BCH is broken for a long time ;)
there was some time we had same problem with BTC B32 addresses so we introduced convert to lower case button
as soon i convert it Suite accepts it.

2.4.3: QR   - BITCOINCASH:QR0QACW7NPWZG7PUTA5XJFM3N8Y24DYSPUZTY804E6 - `Address is not valid`
2.4.3: Suite - bitcoincash:qr0qacw7npwzg7puta5xjfm3n8y24dyspuzty804e6

1.9.4 QR code - BITCOINCASH:QZNG9KY34PGAFJXNRAQTT59JGAA28LH84SPP3R64D4 - `Address is not valid`
1.9.4 Suite - bitcoincash:qzng9ky34pgafjxnraqtt59jgaa28lh84spp3r64d4

[image]

bosomt, 2022-03-11:
2.4.4     revision: 20 bytes 0x0bfdfbe925f64890619325cd0cc7f5c296f491da,
same behavior for BCH address

[image]

bosomt, 2022-03-18:
**QA OK**

retested with latest dev master
sent separate Suite issue to enable uppercase receive address conversion for BCH /other coins like VTC or LTC already enabled/

PR #2164 (2022-03-10): Automatically pick best QR encoding
The `Qr` component will automatically figure out if it can make the QR code smaller ( = bigger pixels) by uppercasing the address.

If not, it will keep lower-case encoding to improve compatibility.

This fixes #1751 because Bech32 Bitcoin addresses fit into the same-size QR code regardless of whether BINARY or ALPHANUMERIC encoding is used.
closed 2022-03-24 by hynek-jina