trezor/trezor-firmware #1761: Download firmwares from different place than wallet.trezor.io in trezorctl

tsusanka, 2021-08-13:
`trezorctl firmware-update` currently downloads firmwares from `https://wallet.trezor.io/data/firmware/X/Y`. As we plan to deprecate wallet.trezor.io (no date set though) we should switch to either:
1. data.trezor.io
2. Suite (but I think Suite fetches it from data.trezor.io)
3. Start using Github releases and fetch from there (probably not worth it)

grdddj, 2021-08-24:
I looked into the current situation, and we are already using `data.trezor.io`, at least in the current `master` firmware.

https://github.com/trezor/trezor-firmware/blob/7aed8d535dc836336d461879e8f7d6c3cdeaf6c9/python/src/trezorlib/cli/firmware.py#L202

The previous `wallet.trezor.io` was replaced in this commit https://github.com/trezor/trezor-firmware/commit/225cd157325dce4d618a8817c2e15d800d4fb9ed on Nov 16, 2020

It is possible that in the released version installable from pip (https://pypi.org/project/trezor/) we still have the `wallet.trezor.io`, but after next release, it should be fine (I see the latest release 2021-07-29, so it may be already there, I will check it)

grdddj, 2021-08-24:
So the latest release of `trezorlib 0.12.3` includes the updated `data.trezor.io` - we are good there

matejcik, 2021-08-24:
thanks for investigating! closing the issue then.

closed 2021-08-24 by matejcik

tsusanka, 2021-08-24:
Ah I indeed had an old 0.12.2 from nix. What a rookie mistake! :)

bosomt, 2021-09-23:
**QA OK**

```
testing@testings-MacBook-Air MacOS % trezorctl firmware-upgrade        
WebUSB transport is disabled: dlopen(libusb-1.0.dylib, 6): image not found
Best available version: 2.4.2
Downloading from https://data.trezor.io/firmware/2/trezor-2.4.2.bin
Trezor T firmware image.

```